guino/BazzDoorbell #59: Mqtt client on firmware 4.0.7?

Rikbruggink, 2021-11-15:
Hi, 

I got root and telnet activated on 4.0.7 version with board

  "devname": "Smart Home Camera",
  "model": "Bell 8S",
  "serialno": "*********",
  "softwareversion": "4.0.7",
  "hardwareversion": "BE8S_A2_V10_433",
  "firmwareversion": "ppstrong-a3-tuya2_general-4.0.7.20210513",

but when i want to start mosquitto_pub from #4 it is core dumping on the sd card. Any pointers how to proceed?

/dev # cat /proc/cpuinfo
Processor       : ARM926EJ-S rev 5 (v5l)
BogoMIPS        : 199.06
Features        : swp half fastmult edsp java
CPU implementer : 0x41
CPU architecture: 5TEJ
CPU variant     : 0x0
CPU part        : 0x926
CPU revision    : 5

Hardware        : CLOUD39EV3_AK3918EV300_MNBD
Revision        : 0000
Serial          : 0000000000000000

guino, 2021-11-15:
@Rikbruggink looks like the mosquitto_pub we have is compiled for arm v7l -- your processor is arm v5l (older) so it is likely crashing because of that.

I am not the person who built/provided the existing mosquitto_pub tool, so your options are:
1-Download sources and compile the mosquitto_pub for arm v5l (using a cross compiler for uclibc)
OR
2-Try to find an alternate mosquitto_pub binary compiled for arm v5l (uclibc)
OR
3-Use an alternate way for notification (i.e. http with wget using the busybox provided)

You could potentially use http+wget (option 3) to any existing computer/device on your network where you have mosquitto_pub by creating a service to run mosquitto_pub to whatever you need.

Rikbruggink, 2021-11-15:
@guino Thanks for the feedback, i figured the pub client was compiled for armv7 found a pull on the Merkury1080P but that also seems for the armv7 toolchain. He/SHE als posted a post script for webhooks. Will look into that. Will let you know the progress here.

pjeterinfo, 2022-01-16:
@Rikbruggink how does your env file look like? I have the same hardware/software revision but doesn't get the device rooted.

guino, 2022-01-16:
@pjeterinfo most of the time when the root process fails is because of the SD card brand/model or format. You should try to format it with the app, linux, windows and see if it helps. You should definitely try other cards but same brand/model will not make much of a difference so you should try different brand/models to see if it helps.

pjeterinfo, 2022-01-17:
[env.zip](https://github.com/guino/BazzDoorbell/files/7883619/env.zip)
@guino see attached zip file.

guino, 2022-01-17:
@pjeterinfo your files look fine -- your issue is most likely due to SD card brand/model or format. Please note the brand/model and format is most important for the boot loader during the rooting process and after that you can probably use the SD cards that are not working right now. It is also worth pushing the reset button during normal operation (no SD card) to make sure it actually does something -- if the reset button isn't working hardware side it won't work during step 5 of https://github.com/guino/BazzDoorbell/issues/13

pjeterinfo, 2022-01-17:
@guino thx for checking, the reset button is working normally. I will try other sd cards.

guino, 2022-01-27:
@kruithofm @Rikbruggink I was able to make this work with homebridge (armv5 compatible):

[mqtt_pub.zip](https://github.com/guino/BazzDoorbell/files/7952908/mqtt_pub.zip)
```
/mnt/mmc01 # ./mqtt_pub 
Usage: mqtt_pub address port topic message
/mnt/mmc01 # ./mqtt_pub a 2 3
Usage: mqtt_pub address port topic message
/mnt/mmc01 # ./mqtt_pub 10.10.10.92 1883 home/doorbell detected

./mqtt_pub disconnecting from 10.10.10.92
```
I did not do a lot of testing but if it works for you let me know and I can post sources and binaries in the repo.

guino, 2022-01-27:
@kruithofm thanks for testing and reporting back.