guino/BazzDoorbell #60: Mini8S camera

vsessink, 2021-11-25:
Hi, I bought one of these cameras at Action.com today. It says it's a "mini8s_a2mb_f37 rev1.1". Same AK3918 chip. 8mbit flash. I extracted the flash - seems to me it looks like the mini9s but I could be wrong.

vsessink, 2021-11-25:
[cameradata.gz](https://github.com/guino/BazzDoorbell/files/7605402/cameradata.gz)
This was extracted from a new (i.e. never used) camera, by desoldering the SPI flash and reading it out.

vsessink, 2021-11-25:
If you need pictures, just let me know. Camera is a bit brain damaged now, without the flash chip, but it looks almost the same.

guino, 2021-11-26:
@vsessink This device should work with the process described here: https://github.com/guino/Merkury1080P#conclusion It seems to have firmware **ppstrong-a3-tuya2_lsc-4.0.6.20210311** and there's a patch for it in https://github.com/guino/ppsapp-rtsp/issues/1 (you can probably try to just modify tuya_config.json to enable onvif without patching).

It doesn't really matter what the hardware is since the firmware in it is one we've seen and it has to be fully compatible for it to work so it should be straight forward in getting it rooted and enabling onvif/rtsp. You will need to register/configure it with the app before you can use onvif/rtsp (the firmware only starts the onfig/rtsp function after it talked with the tuya cloud to sync time, etc), after that it doesn't need to be online and you don't have to use the app if you don't want to use it. You should also be able to make it work 'offline' using the universal offline setup described here: https://github.com/guino/BazzDoorbell/wiki/Can-I-use-the-device-OFFLINE-(disconnected-from-tuya-servers)-%3F

The fact that you got a flash of it before it was ever registered allows you to restore it to factory spec later if you want (and/or downgrade firmware later if you ever want to).

vsessink, 2021-11-26:
Thanks. Still I have a few questions and a remark.
1) I'm seeing a cramfs and jffs2 in my camera block - but where's the rootfs?
2) remark: there are TWO uart connections on this camera. One is easily visible because it has (through hole) interconnects. This one will only show https://github.com/guino/BazzDoorbell/issues/48#issuecomment-908942726 this sort of stuff and is uninterruptable. The other UART is on the front side (i.e. the camera side) of the print and only shows as test pads. _This_ one does the U-Boot-stuff. And it...
3) asks for a password. Any suggestions?

guino, 2021-11-26:
@vsessink
1-The jffs2 filesystem is the rootfs and the cramfs is an application 'partition' they're all in the flash so at chosen addresses so it's not really a partition but it's how they separated it so they can update app, rootfs and boot loader separately if needed.
2-I am assuming one uart is just used for interaction/debugging the application (ppsapp) and the other one (uboot) is used for boot/console if enabled.
3-If you want to mess with uboot stuff on the uart just send me an email privately (it is in my github profile) and I can give you some pointers on that.