guino/BazzDoorbell #63: doorbell v.2.10.5

kakacs, 2021-12-12:
hello @guino  i need help with my doorbell again i have the software 2.10.5 how i have test with your information but i cannt make it run can you help me pls ??

guino, 2021-12-12:
@kakacs I remember you running 2.9.7 -- in fact I still have those files.

1-Did you create a ppsFactoryTool.txt file which is required to enable the URLs in 2.10.x ? (see special note for 2.10.x in #2)
2-If you created ppsFactoryTool.txt what is the response for http://admin:056565099@IP/proc/cmdline from the device ?
3-If you created ppsFactoryTool.txt what is the response for http://admin:056565099@IP/devices/deviceinfo from the device ?
4-Does your SD card still have all the files from 2.9.7 ?

kakacs, 2021-12-12:
hello friend i have make it run. I have create a ppsFactoryTool.txt file which is required to enable the URLs in 2.10.x
and i have put the files fom 2.9.7 on sd card and have rsto stream again . 
this is the reponse from cmdline : mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd - ip=${T//_/$'\\x20'}:::::;T=\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\";eval mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd
and this from deviceinfo: 

devname | "Smart Home Camera"
-- | --
model | "Bell 8S"
serialno | "060665491"
softwareversion | "2.9.7"
hardwareversion | "BE8S_H1_V10_433"
firmwareversion | "ppstrong-c51-tuya2_lcs-2.9.7.20201020"
authkey | "IO013uM1rekxA2HMImdfo2PDz1rB7LpI"
deviceid | "pp01896af2fbff186aa3"
identity | "MR2005212400904064"
pid | "aaa"
WiFi MAC | "d4:d2:d6:d0:2c:bb"
 but one quastion i have it more. why delete my doorbell after a few days or by lost wifi or i restart the doorbell from electric 
the files from the sd card ? have you one idear friend ??

thanks for help