guino/BazzDoorbell #65: Escape chars for ssid in ppsFactoryTool.txt

tmoore22, 2021-12-19:
Hey Guino,

I have a special request because my SSID has a "-" in the label and its really frustrating that the ppsapp cannot deal with it....

Tried most esc chars and using variations of quotes but nothing helps, can you offer any suggestions?

my ssid is: openwrt-r3p

as you can imagine I have nearly 30 devices using this and only the tuya cam (Akaso Bullet 2S) is not capable.

P.s your hack works great, compliments! i recently purchased from Amazon and it has version ppstrong-c51-tuya2_akaso-2.10.0.20210803

guino, 2021-12-19:
@tmoore22 have you tried creating a ppsFactoryTool.txt file in the SD card ? That allows you to set the SSID and Password without any limitations from the app. That information is in #2. 

For 2.10.x you would have to create that file in order to get URLs to work.

tmoore22, 2021-12-19:
yes, all my testing have been by making changes to the ppsFactoryTool.txt ... and reboot the device

"openwrt-r3p", "openwrt\-r3p", 'openwrt-r3p', "openwrt%2dr3p"... tried all different combinations and still fails!

works without the hypen on different ssid (same router)

guino, 2021-12-20:
@tmoore22 I checked the code and I don't see any kind of 'special handling' of the SSID field (in ppsFactoryTool.txt) that would prevent `-` from working in it. It does seem to require the network to be 'visible' (not hidden) because it scans for it before trying to connect to it but otherwise it just gets the SSID in the ppsFactoryTool.txt file and sends it straight up to a connection manager socket which should attempt to connect to the SSID+Password combination. You may be having an issue due to some sort of bug in their code to connect to the Wifi (which could be mishandling the password) or it could just be something else entirely and you may be thinking it is the SSID name.

Only way to be 100% sure would be to change the SSID to something like openwrtr3p (without changing any other settings like password, encryption type, channel, etc) and see if the device connects -- if it connects then you're right and there's likely a bug in the wifi driver when handling `-` in the SSID. If it doesn't connect then the issue is something else and you may be able to figure it out by modifying settings one by one until it connects (or by creating another wifi and testing with it). I'm only suggesting you change the SSID temporarily for testing as I know you said the intention is to avoid having to change the SSID in multiple devices.

tmoore22, 2021-12-20:
I did confirm it works with just: openwrtr3p (no dash) ... so I have confirmed that it is the "-"

do you have any other suggestions of Esc chars that I could try?

Thanks for you replies :) much appreciated!

guino, 2021-12-20:
@tmoore22 If you already tried to place the `-` in the Phone App (to see if it handles it any different) then just about the only thing left you could try is the command `iwconfig` in telnet.

Something like:
`iwconfig wlan0 essid "openwrt-r3p"`

You WILL disconnect after the command is executed and (if it works) you'll have to reconnect again -- you may need to wait a little and try to reconnect to see if it works and if doesn't you'll have to power cycle to get the device connected again. You may need to remove the ppsFactoryTool.txt file before doing the iwconfig command (unsure of this).

I have never tried the above so I'm not really hopeful but it should be safe to test it (the changes will not be persistent).