guino/BazzDoorbell #67: Doorbell ntp timezone is set

stack0423, 2021-12-23:
Where can i change this?

guino, 2021-12-23:
@stack0423 The phone app usually sets the Time Zone for the clock on the device (which is displayed on the video). You can check what it is currently set in the "Device Information" screen under properties of the device in the phone app (i.e. tuya app). Here's some information about the time zone from tuya: https://support.tuya.com/en/help/_detail/K9huti7rwgcou

If you have telnet access the time zone can also be checked in the file /home/cfg/TZ (also in /etc/TZ):
```
/home/cfg # cat TZ
CST05:00:00
```
I have never tried to modify either file to see if it can be used to change the timezone without the app -- there's a chance the app may overwrite it if you change it.

My recommendation would be to follow the information in the tuya link above: set the timezone on the phone, then remove and add the device using the phone (so it takes the timezone from the phone).

stack0423, 2022-01-02:
Sorry , i mean i block doorbell to internet i have local ntp (mikrotik). is that ok for the doorbell to update the time ?

guino, 2022-01-03:
@stack0423 Assuming you're using an offline patch and you blocked access to the tuya servers then you'll need to set the time in the device somehow when it boots -- the most common way is described here: https://github.com/guino/BazzDoorbell/issues/4#issuecomment-744084557 but you could use some custom script/web request in the local network.

bkbartk, 2022-11-07:
for what it's worth,
I just had  the same issue when DST was ending and over a week later the time was still incorrect.
the system time always is CST.
I have internet blocked for my doorbell, but for some reason while I'm on my local LAN the app(LSC/Tuya) still works,
it doesn't when I'm somewhere else.
unfortunately the app didn't fix my timezone, even though I use a local ntp server.
for that reason I granted Internet to my doorbell, rebooted, checked the time, which was ok, and then killed the internet connection again.

In about 6 months while having this issue again, I will check if there is another way of automating the timezone. but now I don't have the time