guino/BazzDoorbell #68: Bell 8S  Software 2.10.5 hack doesn't work

six1, 2021-12-28:
Hi,
i own a Doorbell from Action Market Germany. I want to hack, because i want to use rtsp. 
So far, I tried all, i can found, but no luck at all :-)

I modified ppsFactoryTool.txt, and Port 80 is working.

Files on SD Card:
ppsFactoryTool.txt, env, initrun.sh, ppsMmcTool.txt

http://admin:056565099@192.168.1.6/devices/deviceinfo
`{"devname":"Smart Home Camera","model":"Bell 8S","serialno":"061208572","softwareversion":"2.10.5","hardwareversion":"BE8S_H1_V10_433","firmwareversion":"ppstrong-c51-tuya2_lcs-2.10.5.20210806","authkey":"tl0OfBcSMHd76hUHDoeqL96d60MaZotE","deviceid":"pp01d28dffbde156c4e0","identity":"MR2005212301201042","pid":"aaa","WiFi MAC":"7c:25:da:1b:c9:e9"}`

http://admin:056565099@192.168.1.6/proc/cmdline
`mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd`

http://admin:056565099@192.168.1.6/proc/self/root/etc/init.d/S90PPStrong
`#!/bin/sh

export PATH=/usr/bin:/sbin/:/usr/sbin:/bin

RED="[1;31m"
NORMAL="[0;39m"

echo "${GREEN} 2015 PPStrong Tech Cop.Ltd.${NORMAL}"

mkdir -p /opt/pps
MTDNUM=`cat /proc/cmdline | sed 's/.*ppsAppParts=\([0-9]\).*/\1/'`

# debug
#MTDNUM=5

echo "------------->mtdnum:${MTDNUM}"

case $MTDNUM in
         5)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
			break
			;;
		 7|8)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	break
		 	;;
		 0)     
			sleep 10
			mount -t vfat /dev/mmcblk0p1 /opt/pps
			break
			;;
		 *)
			MTDNUM=5
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	;;
esac

echo "/opt/pps/" > /tmp/PPStrong.runpath
[ -e /opt/pps/initrun.sh ] && cp /opt/pps/initrun.sh /tmp/PPStart && chmod +x /tmp/PPStart && /tmp/PPStart`

I tried to change Sleep Time in initrun.sh from 10 to 30 seconds.

I also tried following to execute hack:
http://admin:056565099@192.168.1.6/proc/self/root/tmp/hack
and  
http://admin:056565099@192.168.1.6/proc/self/root/mnt/mmc01/hack  

but result is always: HTTP ERROR 500

Device can write to sd card, if i enable motion detection. (data files)
What else can i do?

[Doorbell.zip](https://github.com/guino/BazzDoorbell/files/7784102/Doorbell.zip)

guino, 2021-12-28:
@six1 I want to make sure you followed step **5** from #13 -- your files seem ok but it does not seem like the hack was installed correctly. the '.../hack' URLs you posted do not 'install' the hack they only serve to verify if the hack was installed. 

If you followed step 5, there are two possibilities:

1-you could have a defective reset button on the device (which you can try by pressing it to see if it does anything after booting up). A while back I remember a user that was able to apply the hack to one device but not the other despite the fact the reset button appeared to be working.

2-your SD card may not be working correctly (during boot). You could try a different SD card or formatting the SD card (FAT32) with a different tool/computer. I have seen reports of people not being able to apply the hack when the SD card was formatted by the phone app and others that only had success after formatting it in linux. I personally have seen my device not work with some SD cards on boot, but it reads all cards after booting up -- what I am trying to say is: despite the fact that the SD card is working (to record video) it doesn't mean it is working **during boot** which is what we need to apply the hack. After the hack is installed you can use any card you like (even one that did not work to install it).

six1, 2021-12-29:
Hu guino,
thanks for your Answer.
I readed of problems formating sd cards :-) yesterday i ordered 2 new sd cards... we will see.
The only thing i want to change is, to enable rtsp/onvif.
looking at result from: http://admin:056565099@192.168.1.6/proc/self/root/home/cfg/tuya_config.json
`{
	"version":	1,
	"sleep_mode":	0,
	"alarm_fun_onoff":	0,
	"alarm_fun_sensitivity":	1,
	"alarm_fun_mode_switch":	0,
	"alarm_fun_time_start":	0,
	"alarm_fun_time_end":	0,
	"flip_onoff":	0,
	"light_onoff":	1,
	"night_mode":	0,
	"sound_detect_onoff":	0,
	"sound_detect_sensitivity":	0,
	"watermark_onoff":	1,
	"event_record_time":	60,
	"enable_event_record":	2,
	"record_enable":	0,
	"motion_trace":	1,
	"motion_area_switch":	0,
	"motion_area":	"",
	"motion_tracking":	0,
	"cry_detection_switch":	0,
	"humanoid_filter":	1,
	"loudspeaker_vol_pct":	100,
	"jingle_mode":	0,
	"jingle_sound":	1,
	"jingle_volume":	100,
	"jingle_exist":	0,
	"flight_bright_mode":	0,
	"flight_light_brightness":	100,
	"flight_pir_set":	0,
	"flight_pir_one":	0,
	"flight_pir_two":	0,
	"flight_pir_three":	0,
	"flight_pir_sensitivity":	0,
	"flight_alarm_fun_onoff":	1,
	"flight_on_off":	0,
	"flight_pir_light_on_time":	30,
	"flight_warn_switch":	0,
	"flight_dualbrite":	0,
	"flight_ontime":	0,
	"flight_highbrightess":	10,
	"flight_lowbrightess":	10,
	"flight_mode":	0,
	"flight_motion_sens":	0,
	"onvif_enable":	0,
	"onvif_pwd":	"admin"
}`

i want to set "onvif_enable" to "1"
so the sd card is becoming obsolet, if i switched that on :-)

regards, michael

six1, 2021-12-30:
Hi,
i tried another sd card but without success.

What i can see is, that "ppsFactoryTool.txt" is being executed. IP is accessible on Port 80.
"env" is not working correct way. "initrun.sh" is never executet.

Which way "env" is called?

six1, 2021-12-30:
Ok, fixed it with new sd card!
old ones, which doesn't work was Intenso 16/32GB
new one, which works is ScanDisk Ultra

six1, 2021-12-30:
Steps for 2.10.5 "Bell 8S"
`{"devname":"Smart Home Camera","model":"Bell 8S","serialno":"061208572","softwareversion":"2.10.5","hardwareversion":"BE8S_H1_V10_433","firmwareversion":"ppstrong-c51-tuya2_lcs-2.10.5.20210806","authkey":"tl0OfBcSMHd76hUHDoeqL96d60MaZotE","deviceid":"pp01d28dffbde156c4e0","identity":"MR2005212301201042","pid":"aaa","WiFi MAC":"7c:25:da:1b:c9:e9"}`

1) copy env, initrun.sh, ppsMmcTool.txt, ppsFactoryTool.txt to sd card (best will be ScanDisk Ultra)
2) press Reset and hold while inserting Power Cord to Doorbell
3) if "http://admin:056565099@192.168.1.6/proc/self/root/mnt/mmc01/hack" = "done", it's all ok.
4) place sd card into pc  copy files from https://github.com/guino/BazzDoorbell/tree/master/mmc to sd card
5) Edit custom.sh:  "/mnt/mmc01/busybox telnetd"  to  "/mnt/mmc01/busybox telnetd -l /bin/sh"
6) insert card in doorbell and power on
7) 
  login by telnet (assuming you configured it)
  execute cp /home/cfg/tuya_config.json /mnt/mmc01
  remove card, place it on computer, edit line ""onvif_enable":	0," to ""onvif_enable":	1," , save and eject card properly
  insert card on device, wait a few seconds (i.e. 10) so the card is mounted
  execute: cp /mnt/mmc01/tuya_config.json /home/cfg
8) Edit cgi jpeg and mjpeg addresses:
Edit files snap.cgi and mjpeg.cgi in Folder cgi-bin on sd card:
#ADDRESS=42ac2c  <- disable by "#"
ADDRESS=56e6b0  <- insert this line

have fun :-)

guino, 2021-12-30:
@six1 glad you found a working SD card. Now that the hack is installed you CAN most definitely use the device with one of the SD cards that did not work initially (now that the hack is installed) by simply copying the files from the current SD card. For some reason these devices are picky about the SD card when installing the hack but work with most cards after booting up.

ThijsMaas, 2021-12-30:
Hi @six1, I have the same device and firmware as you, and I was wondering if you were able to get rtsp working? I have been able to do the hack and get snap.cgi and mjpeg.cgi working, but not rtsp. I'm not sure if I'm looking on the correct port / url.

six1, 2021-12-30:
Hi Thijs,
this is my path to rtsp:
rtsp://admin:admin@192.168.1.6:8554/Streaming/Channels/101  (or 102 for low res)

Do you execute step 7 from my list? ( "onvif_enable": 1 )

jpeg, mjpeg and rtsp are working like a charme since hours...

six1, 2022-01-02:
The Doorbell is now working since i've "patched" it without any trouble.
I've added the Bell in my Zoneminder Installation over rtsp. Having no trouble.
I ordered a MARMITEK Doorbell on Amazon. It took 2 Sec to pair the Bell and the Gong! Working fine.

URL to Marmitek Gong:
https://www.amazon.de/gp/product/B08CDN13WP/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1

ThijsMaas, 2022-01-03:
I do have onvif working, I have the doorbell streaming using the onvif integration on Home Assistant. I'm just not able to get the rtsp stream working. I have tried to stream it using vlc, but it keeps loading or something. Anyway, I think I don't need that anyway, as I have it working in home assistant now, which was the goal :)

guino, 2022-01-03:
@ThijsMaas ONVIF works using RTSP so it must be working too. Some things to keep in mind: 
1-If you're streaming (ONVIF) on home assistant, you may not be able to stream it (at the same time) somewhere else with RTSP (I don't recall if the onvif implementation has a limit to number of concurrent clients).
2-Some clients (VLC for MAC has been mentioned before) have issues with the video feed from these devices, so trying different clients is always important.
3-Whatever user/password you used with ONVIF is probably what you would need to use with RTSP to make it work.

sinclairfr, 2022-01-19:
Is that normal that RTSP has no sound ?

guino, 2022-01-19:
@sinclairfr all RTSP feeds I have seen have sound. snap/mjpeg on the other hand will not have sound. You may want to try a different player (I recommend VLC).

KapitanKapitan, 2023-02-19:
Any idea why ONVIF doesnt working, but RTSP works perfecly?

I get error: wrong password for ONVIF on my NVR or some software on Windows.

{
	"sn":	"102204378",
	"factory_code":	0,
	"factory_code_str":	"",
	"model":	"Bell 8S",
	"tp":	"lasivvb8ccnsma4t",
	"p2p_uuid":	"",
	"ip":	"192.168.18.156",
	"mask":	"255.255.255.0",
	"gw":	"192.168.18.1",
	"mac":	"b4:fb:e3:d9:ef:65",
	"interface":	"wlan0",
	"version":	"4.0.7"
}

guino, 2023-05-24:
@KapitanKapitan The ONVIF implementation in these devices doesn't work well with a lot of software/NVRs -- You could potentially create your own ONVIF interface using the RTSP stream (There are a number of free applications that can do that).

@aladin2000 If you use #13 : there's no risk of breaking the device / booting without the SD card will boot normally / you can always restore original configuration by simply deleting the files from the SD card.

guino, 2023-05-26:
@aladin2000 you don't need to backup the flash before using #13 -- it is a relatively safe process. If you want to get mqtt doorbell notifications you will need to use the process in #13 then configure the custom.sh file to use the log_parser (to generate mqtt notifications) as described here: https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Use-notifications-MQTT-log_parser-without-a-patched-ppsapp-%3F

The above is probably the quickest solution, and without hacking the device the only way to get bell push notifications is to use the tuya plugin in https://www.scrypted.app/ -- at least I know it can get motion notifications (it may require some actual coding/changes) to get the bell push notifications unless they already implemented it.

guino, 2023-05-27:
@aladin2000 This thread was about firmware 2.10.5 so I thought that was what you had (#13 is only for 2.x firmware)-- I just realized you have firmware 4.0.7 -- for that firmware you need to use the instructions here to root your device: https://github.com/guino/Merkury1080P#conclusion -- you can't use #90 or any of the scripts/tools in these projects without first rooting your device.