guino/BazzDoorbell #69: ieGeek DD201 security camera

youribonnaffe, 2022-01-02:
Hello,

I'm opening this issue to share some of my experiments with an ieGeek DD201 security camera. I found many valuable informations on this repository so I thought it might interest others.

https://github.com/youribonnaffe/iegeek-security-camera

I did not fiddle much with the firmware and rather tried to use the APIs used by the CloudEdge application.
To make it short I changed the app to enable debug logs so I could trace all the API calls made. That allowed me to understand the authentication/signing scheme. 

Unfortunately I did not manage to achieve what I want (enable/disable notifications via API) as logging in clears all the other active sessions.

Thanks to all the contributors for the work put in this repository!

guino, 2022-01-03:
@youribonnaffe you made some interesting progress there. I can tell you that I have seen some firmware code which seems to indicate that when the device is in 'factory mode' (i.e. by presence of ppsFactoryTool.txt or by simply not being setup with the app) a number of features are simply not started including app functionality, cloud connections and video streaming (as you noticed in your research).

I have also seen firmware code that suggests that booting the device without SD card and then inserting the SD card with ppsFactoryTool.txt (AFTER booting) may enable the ports possibly without disabling mentioned features which could be helpful in some ways for your research (it's at least worth a shot).

I wonder if the android application would try to login automatically sometime after being kicked out by another session (maybe wait some time and try again to see if the android app logs back in or not to restore normal operation).

It is highly likely there's a 'test' url (factory mode) that would return a jpeg from the camera image, or maybe even an actual stream (it just depends on the firmware), the only real way of knowing would be to get a copy of the firmware (with a hardware programmer or maybe by UART access) or at the very least find some upgrade file that from which we can extract the main application.

youribonnaffe, 2022-01-08:
Thanks @guino 

Booting the camera without the SD card and then inserting it with the ppsFactoryTool.txt works! I can keep the camera working and have a local port open!

By checking how the app upgrades the firmware I found the HTTP request made to check if upgrades are available and by changing the version I was able to retrieve a upgrade bin file.

https://apis-eu-frankfurt.cloudedge360.com/v3/app/device/upgrade/version/get
-> http://47.91.95.246/firmware/ppstrong-b8-neutral_std-2.3.0.20210928-upgrade.bin

I tried to open it with Ghidra but no luck so far (I'm not familiar with the tool at all)

guino, 2022-01-08:
@youribonnaffe The firmware update is not the same as the firmware itself but we may be able to extract the the main application from it so so we have more details about it. I will try to download and run it thru binwalk to see what happens.

youribonnaffe, 2022-01-08:
Yeah the file looked small and given the name it has...

I made progress on enabling/disabling the notifications though. The API key is printed in the logs when running the application (the original one), so with that I can build queries to enable/disable the notifications for my phone (and without logging elsewhere). I'm not sure how long the API key is valid though. And then I still have to check if this can be done with my spouse's phone (iPhone) so I can do it for her too.

youribonnaffe, 2022-01-09:
Also with an iPhone, setting up mitm is much easier and the CloudEdge app does not do certificate pinning so we can seen all the API calls made.

I'm also trying to parse https://github.com/Mearitek/MeariSdk/tree/master/Android/library/x86_64 with Ghidra now. I'd like to figure out how the p2p protocol works. Ideally if I can remove wake the device with the API and then use this udp port it opens locally that would be cool.

youribonnaffe, 2022-01-09:
I've updated the page on https://github.com/youribonnaffe/iegeek-security-camera to reflect how I managed to make it work. It turns the API signature is not really checked, so the most important thing is the user key that you can retrieve from the application logs in Android Studio. I'm not sure if it'll stay valid for a long time (but I could not find any refresh logic in the Android app code).

guino, 2022-01-10:
Nice job. Sorry I have not had a chance to look at the firmware update file yet, but it sounds like you don’t really need it anymore.

guino, 2022-01-10:
@youribonnaffe the upgrade file you provided seemed to only have the kernel and partial rootfs in it (this is normal for some updates) -- I'm not sure if the URL you used has a parameter for the current version (in which case you could try a different version to see if you get a different upgrade file (hopefully one that contains more data).

I assume you did not get the device to respond to /proc/cmdline ? there's really not a lot we can do without the device or a firmware dump (which you could try to get using #11 or this https://github.com/guino/BazzDoorbell/issues/35#issuecomment-861362737). Obviously if you're satisfied with what you have working there's no need to spend more time on it.

youribonnaffe, 2022-01-10:
Yeah I tried a few different URLs but had not luck with that. The /proc/cmdline was not working either.

StefanoGiu, 2023-10-13:
Hello, great job! I'm trying to integrate a DEATTI Battery Webcam that uses cloudedge in Home Assistant. I would like to create a custom component to have the camera available and visible in HA. Can you share how to use the APIs of CloudEdge please?

timre1337, 2025-02-10:
Is there still activity to this? :)

youribonnaffe, 2025-02-11:
I don't own this camera anymore so no longer tinkering with it on my side.

timre1337, 2025-02-11:
Thanks for your info!