guino/BazzDoorbell #70: Bell 8s: Reboot by command possible?

six1, 2022-01-02:
Hi,
is it possible, to reboot (restart) the Doorbell by command?
i.e.:  "http://admin:056565099@192.168.1.6/proc/restart"

best regards, michael

guino, 2022-01-03:
@six1 As posted here: https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Upload-files-on-your-camera-remotely-using-your-Web-Browser. You can use the following link to soft-reboot the device:

http://admin:056565099@IP/devices/reboot

Please notice that if your device is frozen or partially frozen at a hardware level the above may work and the device will reboot but whatever is frozen hardware side will probably remain frozen/partially frozen. The point of this note is that I have a device where the video stops streaming (app and rtsp) from time to time and the above link is NOT enough to fix the issue when it happens (a full power cycle is required to fix it -- sometimes more than one power cycle is required to fix it).

six1, 2022-01-03:
works like a charme! :-)  i saw it elsewhere but didn't found it anymore...
Thanks guino!