guino/BazzDoorbell #71: Bell 8S busybox

kruithofm, 2022-01-25:
Hi,
I have a smart doorbel camera from the dutch Action
"devname":"Smart Home Camera"
"model":"Bell 8S"
"serialno":"101531256"
"softwareversion":"4.0.7"
"hardwareversion":"BE8S_A2_V10_433"
"firmwareversion":"ppstrong-a3-tuya2_general-4.0.7.20210513"

I tried to get a telnet connection somhow, but it just does not work.
In custom.sh I did put:
/mnt/mmc01/busybox telnetd -l /bin/sh
/mnt/mmc01/busybox httpd -c /mnt/mmc01/httpd.conf -h /mnt/mmc01 -p 8080
/mnt/mmc01/busybox > /mnt/mmc01/busybox.log

I also did put 2>&1 >> /mnt/mmc01/telnet.log to check telnet on errors, no result.
All files are created, only they are empty. (http://admin:admin@ip:8090/proc/self/root/mnt/mmc01/busybox.log)
Port 8080 and port 23 are closed.
So to me it looks like busybox is not working, (Merkury720-main, Doorbel_BazzDoorbel, and more)
I also tried 3 different sd-cards with the same result.
I downloaded serveral busybox files. without results.

Anyone any adea?
Thanks

guino, 2022-01-25:
@kruithofm your busybox must not be working and it is fairly common to download it incorrectly. The busybox file should be about 1Mb in size (if you download it incorrectly it will be fairly small). Here's a link for the correct busybox to download: https://github.com/guino/BazzDoorbell/blob/master/mmc/busybox?raw=true

If you have issues with the above busybox you can also try this one:
https://github.com/guino/Merkury720/blob/main/mmc/busybox?raw=true

guino, 2022-01-25:
I also recommend making sure you 'eject' the SD card properly from the computer after copying the files to avoid file corruption.

kruithofm, 2022-01-25:
Thanks for the response. I always eject the card properly (Learned the hard way in the past :-) ). The file is always (from all sources I download it from) around 1.1 Mb. Even on the SD-card I checked. I also binairy compared the file with what I downloaded after I tried it to be sure its correct. 
So I'm pritty shure that the file should be oke. Still no luck.

kruithofm, 2022-01-25:
Pffff... Oke, This one is solved... I did a factory reset (buton pused for 5 seconds when it starts) and now everything is working.

closed 2022-01-25 by kruithofm