guino/BazzDoorbell #72: Cleverio CD-100 Doorbell working?

labaland, 2022-02-11:
Is it possible on this doorbell camera?
Cleverio CD-100 [https://www.kjell.com/globalassets/productimages/825612_51495_7.tif?ref=5A6A8C178C&format=jpg&w=960&h=960&mode=pad](url)

guino, 2022-02-11:
I have not heard anyone ever using this device but if it uses the tuya app it is possible for it to work.

Bottom line is:
-If you already have the device it is worth trying (should not damage it if it doesn’t work)
-If you don’t have the device I would look for a device that has been confirmed to be working.

labaland, 2022-02-11:
> 

Than you for super fast answer! i have seen the same doorbell but with diffent brand name i guess (cleverio) swedish brand? 
Can u please sen me the link for the file and guide i need to try this? im not so good with theese things but i want to try :D Doorbell was only like 36$ so :D Thank you :)

labaland, 2022-02-11:
> I have not heard anyone ever using this device but if it uses the tuya app it is possible for it to work.
>
> Bottom line is: -If you already have the device it is worth trying (should not damage it if [...]

> 

Than you for super fast answer! i have seen the same doorbell but with diffent brand name i guess (cleverio) swedish brand? 
Can u please sen me the link for the file and guide i need to try this? im not so good with theese things but i want to try :D Doorbell was only like 36$ so :D Thank you :)

guino, 2022-02-11:
@labaland since I know nothing about your device the only guide I can suggest you follow is this one: https://github.com/guino/BazzDoorbell/issues/2 -- depending on what firmware version you find or responses you get from the device you'll be linked to different guides (or won't be able to do anything if you can't get the information from the device).