guino/BazzDoorbell #73: Think I have killed it

jas0npc, 2022-02-20:
@guino Help, I think I have killed my iegeek Bell J5

{"devname":"Smart Home Camera","model":"Bell 5T","serialno":"063726273","softwareversion":"4.3.3","hardwareversion":"BE5T_H1_V10_433","firmwareversion":"ppstrong-c5-s_joys-4.3.3.20210122","identity":"MR2008100401612516","uuid":"","licence_id":"ppsccabb53a7e05d47c2","WiFi MAC":"7c:25:da:a4:f9:0f"}

First responce from step 1.
mem=36M console=ttyAMA0,115200n8 mtdparts=hi_sfc:384k(bld)ro,64k(env),64k(enc)ro,64k(sysflg),3584k(sys),6656k(app),1536k(cfg),1m(recove),2880k(user),128k(oeminfo) ppsAppParts=5 ppsWatchInitEnd

last env i used before bell refused to boot.

bootargs=mem=36 console=ttyAMA0,115200n8 mtdparts=hi_sfc:384k(bld)ro,64k(env),64k(enc)ro,64k(sysflg),3584k(sys),6656k(app),1536k(cfg),1m(recove),2880k(user),128k(oeminfo) ppsAppParts=0 ppsWatchInitEnd ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYo

Hxd verified i had 0x0A 0x00 at the end,

env file on sd card now
bootargs=mem=36M console=ttyAMA0,115200n8

files on sdcard
env, ppsMmcTool.txt

Is all hope lost, I stupidly did not do a backup of the fw using step 11,

Is their anything I can do to save it?

guino, 2022-02-20:
@jas0npc you can’t really ‘kill’ the device by trying to install this hack - the boot loader should still work to restore the original settings you backed up with step 1. 

All you have to do is follow the troubleshooting/restore steps at the bottom of the post  on https://github.com/guino/BazzDoorbell/issues/2 and it should restore your original settings.

If for some reason it doesn’t work, feel free to post a zip of your restore env+ppsMmcTool.txt file so I can review it. If the camera doesn’t show any signs of life (no lights come on at all) then it is likely just a coincidence that your hardware failed (power issue, board issue , some stuck button or issue with cable or power supply).

jas0npc, 2022-02-20:
@guino Many thanks forr your reasuring words, I have tried again, all the device does is light up red, Every now and again the red bull push button briefly goes off and comes back on again. I have attached the requested zip file. Any help apprectated, All i was trying to do was to enable the RTSP streams to feed into motioneye.

Thank you
[G.zip](https://github.com/guino/BazzDoorbell/files/8104595/G.zip)

guino, 2022-02-20:
@jas0npc your files look fine -- when you power up with the reset button pressed do you see the light blink/change momentarily at all or does it just stay red ? If the SD card files are recognized I believe the light should briefly change (compared to what it does when booting without pressing reset). You should definitely try different SD cards and different format tools to see if you can get it restored. 

Did you by any chance format/or modify the SD card at all after attempting to root it ? The old method (which you used) requires the SD card to be present with a valid format and the correct initrun.sh in it so the device can boot up (without it the light will just stay red), so please try to also keep initrun.sh in the SD card whenever you try different SD cards and/or format tools.

jas0npc, 2022-02-21:
@guino Thank you my friend, That has worked, So from my understanding now, Is the camera back to factory?, So I start again

guino, 2022-02-21:
@jas0npc glad you got it working again, and yes: the device should be in factory state now.

if you’re going to try again please use #13 which is safer and allows the device to boot even without SD card.

jas0npc, 2022-02-21:
@guino all rooted, thank you, Would you be able to check ppsapp and patch it please?
[ppsapp.zip](https://github.com/guino/BazzDoorbell/files/8111516/ppsapp.zip)

guino, 2022-02-21:
@jas0npc which patch do you need ? versions 4.x should allow enabling onvif/rtsp by simply modifying the config json file: https://github.com/guino/Merkury1080P/issues/9#issuecomment-926414826

If you want to run the device offline I ask that you try the 'universal' offline method first:
https://github.com/guino/ppsapp-rtsp/issues/21#issuecomment-918777873

jas0npc, 2022-02-21:
@guino I have tried modifying the file but i cant get access to the rtsp streams

guino, 2022-02-21:
@jas0npc I don't believe I have had a single confirmation that a patch I made for 4.3.3 has worked. I can try but I won't guarantee anything.

jas0npc, 2022-02-21:
Sorry, I meant i Changed the cfg file, rebooted, as im using vlc i cant tell if rtsp works.

jas0npc, 2022-02-22:
Hi mate, I'm prepared to be the tester, would it be possible to roll back
the firmware with correct hardware flasher and firmware ?

guino, 2022-02-22:
@jas0npc if you flash a different firmware it must be for matching hardware (BE5T_H1_V10_433) -- I do **not** have this firmware, in fact I only have 1-2 firmware files other than the firmware for my own devices. **Additionally** whoever provides the firmware file has to trust that you'll only use the device offline or the original device won't be able to connect to tuya servers anymore as there will be multiple devices trying to connect to the server with the same ID/license information.

If you know someone who has the same device and you can tell them that you will only use the device offline then you may be able to flash it and use it offline, but honestly, just give me some time and I'll try to make the patch for you.

jas0npc, 2022-02-23:
@guino, Many thanks my friend, I appriceate all the help you have given me and others so far.

guino, 2022-02-24:
@jas0npc  you can try this patch:
[ppsapp-onvif.zip](https://github.com/guino/BazzDoorbell/files/8131589/ppsapp-onvif.zip) use these steps: https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080

jpeg address: 04802b0
play address: 0483b94
onvif password is probably (unless you change it in the config json file): YWRtaW4=

jas0npc, 2022-02-27:
@guino Many thanks for all that you have done, Im going to restart my laptop, Restart the doorbell from factory, then start again, I will come come back and let you know when it works. Thank you

guino, 2022-02-27:
@jas0npc if the patched causes a boot loop, please try the original ppsapp (no changes) on the root of the SD card to see if it also causes a boot loop (depending on the hardware/drivers this may require changes to the rooting process).