guino/BazzDoorbell #74: Also killed my Doorbell, but restoring won't work.

sanscorp1, 2022-03-01:
I have two doorbells, one stock and one none functional at the moment.
Both are identical (SW and HW).

I am afraid I used the wrong env and don't have a backup of the cmdline anymore.
The camera is version 4.0.7 and I first need to restore it back to factory default.

I now have telnet again, but the device won't register to the Android Tuya app.
Is there anything I can do from the CLI?

I set env to:
bootargs=mem=64M console=ttyAYA0,115200n8

guino, 2022-03-03:
@sanscorp1 sorry for the delay, I haven't had a chance to hop back into github lately. 

The cmdline you showed me is from #13 (good) -- that cmdline should allow the device to boot even without a SD card installed, so if the device isn't starting up normally/stock without the SD card installed then it could only be due to hardware malfunction (which would be unrelated to the rooting process). 

If the device is booting and allowing you to telnet into it (which can only happen with SD card in place): it may not register to the tuya cloud/app if ppsFactoryTool.txt is present in the SD card -- this doesn't happen on all firmware but it does happen on some firmware versions. Just rename the ppsFactoryTool.txt file (to anything else like disabled-ppsFactoryTool.txt) and reboot to see if the device works normally (the built-in URLs like /proc/cmdline won't work anymore but they are only needed for rooting which you have completed since you have telnet).

sanscorp1, 2022-03-03:
OK maybe I cheered to soon.
After a reboot, no more telnet, no RTSP, no connection to the Tuya app.

Removing the SD card and resetting also results in not being able to reconnect as a new device.

I included the contents of the SD card last used:
[Copy SD card Doorbell.zip](https://github.com/guino/BazzDoorbell/files/8178609/Copy.SD.card.Doorbell.zip)

guino, 2022-03-03:
@sanscorp1 if it is not booting without the SD card, then either you messed up the /proc/cmdline when restoring it OR it is a hardware malfunction.

If both doorbells are the same model/firmware, what I recommend you do is:
1. Create a ppsFactoryTool.txt for the working/stock doorbell
2. Get the /proc/cmdline from the working doorbell (only requires ppsFactoryTool.txt -- no need to modify/root it). Once you get the 'original' /proc/cmdline you can delete the ppsFactoryTool.txt file and keep it stock/original.
3. Using the 'original' /proc/cmdline from the working doorbell, use the troubleshoot/restore process from #2 to create the env file and restore the device but **BE SURE** to use the ppsMmcTool.txt file from Merkury1080P since you posted your firmware is 4.x

Doing that should restore the boot settings to what they were originally and allow the device to boot stock -- if this doesn't happen you're likely dealing with a hardware issue in which case I'd recommend claiming warranty if it is possible.

sanscorp1, 2022-03-03:
For the ease of things I have used the USB connector with a standard Samsung USB charger.
It would seem this has caused it's own problems.

After a reboot with the correct env file I am able to reconnect to the Tuya app.
Copying the known working SD card from the indoor camera, offline.log is also filled.

The only question now is will it keep working or is the HW indeed malfunctioning.
Will keep you posted!

guino, 2022-03-03:
@sanscorp1 check your 'samsung' charger -- you have to be really careful cause a lot of those output more than 5V and can damage 5V devices that are not ready for these 'fast' chargers. I have burnt a Raspberry pi board with one of those samsung chargers in the past, so you may have damaged it. It would be too perfect if all chargers with USB plugs were 5V -- samsung logic (and many others) for you.

sanscorp1, 2022-03-03:
I will check the consistency of reboots, if inconsistent I will return it for warrenty.

The second one should be a matter of plugin in the SD card and it's good to go.

sanscorp1, 2022-03-03:
The other (known good / stock camera) shows exactly the same behaviour when inserting the SD card and booting with the reset button pressed for 5 seconds (the first time).

The current SD card is a 1:1 copy of the known working Indoor camera, with the same Version 4.x.x hack.

Booted the camera 4 times, 3 times nothing, no RTSP, telnet or tuya app.
The 4th time everything is working.

Beats me.

guino, 2022-03-03:
@sanscorp1 are you testing with the offline script ? if so the time it takes to re-enable the wifi may vary on each boot. I would be sure to test without the offline script and no ppsapp file in the root of the SD card to see if you get more consistent results.

The other thing to notice is the first boot may take a little longer as it decompresses the home/* files from the firmware.

You could also test if the devices consistently boot normally without the SD card (despite the cmdline being different after rooting).

sanscorp1, 2022-03-03:
Yes the offline script is running.
I will test better and with difference scenario's over a larger amount of time.