guino/BazzDoorbell #77: What to do when your sdcard got full and the root/boot files are gone

sinclairfr, 2022-04-21:
Hello,
I got unlucky and let the sccard that was in  my Bell 8S door bell get full.
All files are gone, the device won't boot anymore.
So far I tried the first steps : 

- putting only the ppsFactoryTool.txt file => no boot and no connectivity on the network
- adding env, initrun.sh, ppsMmcTool.txt and holding the reset button while turning on the device => the lights turns blue and red but still no boot on release
- putting all the files (cgi-bin etc) => no boot

The device is not bricked but I was wondering what I have to do to make it work again.

Any help will be greatly appreciated.

Thanks

guino, 2022-04-21:
@sinclairfr In theory all you have to do is put back the files (except for ppsFactoryTool.txt which is only used to get the initial information).

You seem to have indicated you did the boot with ppsMmcTool.txt and reset button **again** -- this could have installed the wrong version of the hack (you didn't say what firmware you have) if you had the wrong files in place.

Unless you used the hack from #2 (obsolete) the device should always boot, even without a SD card.

If you don't have a backup of the files that were working, I would suggest this:
-Leave only ppsFactoryTool.txt on the SD card (you have to make the information in the file is correct for your Wifi to connect)
-Boot normally (it may not work with the app but it should allow you to query URLs like /proc/cmdline and /devices/deviceinfo) -- you should get the responses from these and post them here so we know what is the firmware and what is currently configured.
-Remove ppsFactoryTool.txt (or the SD card entirely) to use the device with the app while you don't have the correct files.

Based on the URL responses I should be able to tell you what files you need to get it working.

sinclairfr, 2022-04-21:
I really appreciate you taking the time to give me such a detailed answer. Thank you for that :)

To give you more details : 

- I did use the #2 method, I did not know it was obsolete. Quite frankly I don't recall the firmware I used.
- My board looks a lot like the one displayed in the Readme file of this repo. It says Bell5S-HIMB GC2063 REV1_1. It is an LSC Doorbell bought at an Action store.
- Yet, for some reason, I remember that the `deviceinfo` command told me the board was a Bell 8S, not a 5S. 
- The bug came from me realizing that the SD was almost full and accessing the doorbell via telnet and doing a dumb `rm -rf /mnt` to clear the recordings.

I will hook up a serial reader on the board to see what's going on during the boot, I will post whatever piece of information I find in a next message. I will also try your suggestion first.

Thanks again

guino, 2022-04-21:
@sinclairfr if you used #2 and the hack is currently installed it will only boot if the initrun.sh file is on the SD card - just download it from #2 and make sure the card is fat32 formatted before copying it then try to boot normally - I would expect it to work. To get RTSP working you will need to patch ppsapp again and copy it to the SD card along with custom.sh like before.

let me know if that works

sinclairfr, 2022-04-21:
With only the `initrun.sh ` file on an fat32 formatted sd card, the button remains red and nothing happens.
Same result by holding the reset button while powering. 
That's strange.

sinclairfr, 2022-04-21:
I think I know what's going on here : since I did a `rm -f /mnt` when I attempted to clear the recordings maybe I deleted important files for the hack.

guino, 2022-04-21:
@sinclairfr booting with reset pressed will only have any effect if both ppsMmcTool.txt and env files are in place -- Please note: the env file for #2 **must** be adjusted to your or original /proc/cmdline or it won't work. If you remember (or posted) your original /proc/cmdline you should try the Troubleshooting steps in #2 to restore your device to original settings, then (if it boots normally) you can use #13 to hack it again. If you don't remember your original /proc/cmdline I suppose you can try the one posted as example but I would recommend you do your best to use the original /proc/cmdline.

sinclairfr, 2022-04-21:
Wow, I got lucky and found my old env file using TimeMachine on my mac. The devices boots again.
It turns out I am using the 2.1 as per the deviceinfo dump 👍 
`{
  "devname": "Smart Home Camera",
  "model": "Bell 8S",
  "serialno": "060668305",
  "softwareversion": "2.10.5",
  "hardwareversion": "BE8S_H1_V10_433",
  "firmwareversion": "ppstrong-c51-tuya2_lcs-2.10.5.20210806",
  "authkey": "yc1myCBtYK5wJK2dbLpmgW9hqZDuBju2",
  "deviceid": "XXX",
  "identity": "XXX",
  "pid": "aaa",
  "WiFi MAC": "XXX"
}`
I think the #13 method might not be suited for my firmware. I guess I'll stick with the gold old #2 for now.

Again, thank you for your great help @guino .

guino, 2022-04-21:
@sinclairfr glad you found your old files and got it working again -- always keep a backup handy. #13 should work with your device too but if it is working there's no need to make changes. The only benefit of #13 is allowing the the device to boot without SD card.