guino/BazzDoorbell #80: Hello, can you please help me, I can't get any further.  I have this model.

guino, 2022-07-21:
Hello, can you please help me, I can't get any further.  I have this model.
[image]

This comes out when I enter this url http://admin:056565099@192.168.x.x/proc/cmdline

`console=/dev/null LX_MEM=0x3fe0000 mma_heap=mma_heap_name0,miu=0,sz=0x1d00000 pcbversion=B3S_S1_V10 sensor=gc2063mipi`

What should I do next?  My English isn't that good either.

_Originally posted by @savas20041 in https://github.com/guino/BazzDoorbell/issues/2#issuecomment-1186149831_

guino, 2022-07-21:
@savas20041 There's no solution for firmware 5.0.5 as far as I know -- from what I have seen this firmware runs RTOS (not linux) so it won't work with anything we have posted here.

bkbartk, 2022-08-03:
TLDR: would the UART possibility work for this the device? or shouldn't I even try?

my full question:

that's bad news,
I also have a device running firmware 5.
```
{
    "devname": "Smart Home Camera",
    "model": "Bell 8S",
    "serialno": "104077366",
    "softwareversion": "5.2.2",
    "hardwareversion": "BE8S_A2_V10_433",
    "firmwareversion": "ppstrong-a3-tuya2_general-5.2.2.20210914",
    "identity": "M4L0135C7L00908286",
    "authkey": "iL3NmAy8ymgztYNhaFw3Co1LIxZqGt46",
    "deviceid": "tuya629519ecb070e6dc",
    "pid": "aaa",
    "WiFi MAC": "84:7a:b6:ed:dd:c5",
    "ETH MAC": "84:7a:b6:ed:dd:c5"
}
```

and cmdline is like
```
setenv bootargs mem=64M console=ttySAK0,115200n8 loglevel=10   mtdparts=spi0.0:256k(bld),64k(env),64k(enc),64k(sysflg),3m(sys),4032k(app),640k(cfg) ppsAppParts=5 ip=192.168.1.99:::255.255.255.0 eth=00:55:7b:b5:7d:f7 ethDisable usb_other
```

which already contains the bootargs but in a different way so I was confused.
my user credentials btw are admin:admin

anyway, would the UART possibility work for this the device? or shouldn't I even try?

If this is possible I probably will have some additional questions, like how to disconnect pin6 and should I connect pin1 to ground, just like ESP modules.

Thank you in advance for taking the time to answer my question.

guino, 2022-08-03:
@bkbartk based on the provided cmdline there is a good chance the process for Merkury 1080 may work on your 5.2.2 firmware: https://github.com/guino/Merkury1080P#conclusion

The above doesn’t require you to manually edit any boot parameter as it should preserve the original boot parameters by just modifying what it needs. If it doesn’t work it is unlikely to cause any damage either.

For future reference the mtdparts= parameter in the cmdline is usually a sign the device is running linux (and can be rooted). 

I would try that first, if that doesn’t work UART may be an option if you have the will to go that route.

bkbartk, 2022-08-04:
thanks,

I did some testing and after some Merkury1080P changes the cmdline  suddenly looked way more as the manual,
so I started the BazzDoorbell manual again, as I still have a doorbell.(this one: https://www.smarthomeweb.nl/lsc-smart-connect-video-doorbell-review/)

the "hack" page says "done" so that's ok, but I think I have some problems using bussybox.
All pages are parsed as plain text.
[image]
also index.html, which I assume shouldn't,
when I check my sd(16GB formatted by windows as Fat32) it didn't create the direcotories home and lib, so I can't try ppsapp patch.

but a lot of bussybox files are added,

[image]
sorry for taking up your time, but can you point me in some direction?

guino, 2022-08-04:
@bkbartk your device is definitely rooted. The busybox you're using may need to be switched.. there's one for armv5 and one for armv7 -- please delete the core-busybox files from the SD card and try this busybox instead: https://github.com/guino/Merkury720/blob/main/mmc/busybox?raw=true

The URLs you should use for index.html should be: http://user:password@192.168.180.21:8080/index.html -- port 8090 is for the ppsapp URL.
Also make sure your initrun.sh is this one: https://github.com/guino/Merkury1080P/blob/main/mmc/initrun.sh (otherwise you may not get the home directory created.

bkbartk, 2022-08-05:
Thanks,

That works great, I got RSTP working using ONVIF, and it's really nice.
Still index.html only shows a white page and snap isn't working.
I think I have the wrong address,

I tried ghidra, but actually I really don't have a clue what to look for. My MD5 unfortunately isn't in list
[image]

Also the factory firmware is also running, this might be as designed, but is there a way to stop this? Eventually it all will work I don't want to have anything going to the cloud.

guino, 2022-08-05:
@bkbartk snap/mjpeg and play.cgi require the exact address (specific to your ppsapp file), if you post a zip of your ppsapp I can find that out for you.

The same application that runs the RTSP/ONVIF server runs the cloud functions so the only way to disable the cloud is to block the internet from the device after it starts up (it still requires internet to boot up, get the time, etc). There's information on how to get the device offline here: https://github.com/guino/BazzDoorbell/wiki/Can-I-use-the-device-OFFLINE-(disconnected-from-tuya-servers)-%3F

bkbartk, 2022-08-05:
[ppsapp.zip](https://github.com/guino/BazzDoorbell/files/9271614/ppsapp.zip)
That would be really appreciated,

for time, at the moment I run my own ntp server, so that might helps me, but maybe it also needs some other communication, I will read that later, for now, I'm trying to get MQTT working first.

guino, 2022-08-05:
@bkbartk This should be your mjpeg/snap.cgi address: 0x0304c7c
play.cgi address: 0x0305a7c

You can try this patch to make the device start offline (no internet), but you will need to disable the internet on it somehow (router/firewall/hosts file) and you'll need to use your ntp server to set the date/time on the device:
[ppsapp-offline.zip](https://github.com/guino/BazzDoorbell/files/9273584/ppsapp-offline.zip)
Please use this guide to patch it: [HOW TO PATCH](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)

bkbartk, 2022-08-06:
thnx, snap ans mjpeg are working,
really great,
play.cgi "does not exist!" so probably not available on my device

I will try the offline patch later.

guino, 2022-08-06:
@bkbartk for play.cgi you need to provide parameters, see: https://github.com/guino/BazzDoorbell/issues/10#issuecomment-748513226

bkbartk, 2022-08-06:
nice, really cool

savas20041, 2022-08-09:
could that also work with a camera?

guino, 2022-08-09:
@savas20041 if you’re talking about play.cgi - it should work on any device that can play sounds (bell, camera, etc)

savas20041, 2022-08-09:
I meant to root it with this method so that I can get the stream url.  The rtsp link.

guino, 2022-08-11:
@savas20041 there are rooting methods for devices (doorbell and cameras) on different versions: depending on your device (camera) firmware you may be able to root it to enable:
2.7.x: https://github.com/guino/Merkury720
2.9.x : https://github.com/guino/BazzDoorbell/issues/2
4.x (and some 5.x) : https://github.com/guino/Merkury1080P

It doesn't matter if the device is a camera or a doorbell (the process is the same)