guino/BazzDoorbell #82: 4.0.7: solution(s) for my device

jonesMeUp, 2022-08-10:
i have a LSC-Bell 8S (BE8S_A2_V10_433 , ppstrong-a3-tuya2_general-4.0.7.20210513) 
i use the device with **offline option 1** and not all of your scripts worked for me.
as a dump linux user, i invested a huge amount of time to get my problems solved and want to share my solution to help other people with same problems.
- i put all ppsapp messages to /dev/kmsg (i tested **some** of the log_parser.sh posted here, without success as the loop there freezed my rtsp stream)
-  i made a cron job to check the size of the stream folder twice a day and delete folders step by step (sorted by date and am/pm)
- the original custom.h didn't find the ppsapp, as 4.0.7 didn't have a modified one in /mnt/mmc01/ so it made some trouble because the kill cmd was never executed. as there was some more trouble i had to rewrite custum.sh anyway and didn't just copy the file.
- no endless loop in offline.sh anymore
- changing telenet passwd now works for me (passwd -a des <new passwd>)

i would appreciate every improvement from skilled linux guys...

[custom.sh]
```
#!/bin/sh

if [ ! -e /tmp/customrun ]; then
  echo custom > /tmp/customrun
  #####################################################################################
  # Update password to user setting
  # (change password via 'passwd -a des <new passwd>')
  #####################################################################################
  cp /mnt/mmc01/passwd /etc/passwd
  cp /mnt/mmc01/shadow /etc/shadow
  #####################################################################################
  # Start ssh server (create key is needed only once)
  #####################################################################################
  mkdir /etc/dropbear
  ln -s /mnt/mmc01/dropbearmulti /bin/scp
  #/mnt/mmc01/dropbearmulti dropbearkey -t rsa -f /etc/dropbear/dropbear_rsa_host_key
  cp /mnt/mmc01/dropbear_rsa_host_key /etc/dropbear/
  /mnt/mmc01/dropbearmulti dropbear
  #####################################################################################
  # Start telnet and httpd
  #####################################################################################
  /mnt/mmc01/busybox telnetd
  /mnt/mmc01/busybox httpd -c /mnt/mmc01/httpd.conf -h /mnt/mmc01 -p 8080
  #####################################################################################
  # Restart ppsapp to forward the event messages to /dev/kmsg
  #####################################################################################
  PPSID=$(ps | grep -v grep | grep ppsapp | awk '{print $1}')
  #echo "ppsapp id: $PPSID"
  kill $PPSID
  sleep 2 # kill command needs 2 seconds 
  #/mnt/mmc01/set record_enable 0
  #/mnt/mmc01/set enable_event_record 1
  #/mnt/mmc01/set onvif_enable 1
  /mnt/mmc01/home/app/ppsapp > /dev/kmsg &
  /mnt/mmc01/log_parser.sh &
  #####################################################################################
  # Put the doorbell offline
  #####################################################################################
  /mnt/mmc01/offline.sh &
  #####################################################################################
  # Start cron job to delete videos before sd-card is full
  #####################################################################################
  mkdir -p /var/spool/cron/crontabs > /dev/null
  echo "0 7,19 * * * /mnt/mmc01/cleanup.sh" > /var/spool/cron/crontabs/admin
  /mnt/mmc01/busybox crond
fi
```

[log_parser.sh]
```
#!/bin/sh

#####################################################################################
# Doorbell events are send to /dev/kmsg (in custom.sh)
#####################################################################################
# - edit grep cmd for your device to find the right msg
# - edit mqtt cmd for your environment
#####################################################################################
IFS='$\n' # Internal Field Separator
dmesg -c > /dev/null # clear buffer
while true; do 
  BUF=$(dmesg -c | grep 'door bell\|motion' | grep -v 'ERR')
  ###################################################################################
  # motion event
  ###################################################################################
  if test "${BUF#*motion detected}" != "$BUF"; then
	#echo -e "\n$BUF"
	/mnt/mmc01/mqtt_pub 192.168.178.100 1883 home/doorbell "motion detected"
  fi
  ###################################################################################
  # doorbell event:
  # $BUF can hold >1 row, so up/down events are mostly 
  # in the same chunk. so we can use this simplification
  ###################################################################################
  if test "${BUF#*<0>door bell  press down}" != "$BUF"; then
    #echo -e "\n$BUF"
	/mnt/mmc01/mqtt_pub 192.168.178.100 1883 home/doorbell "doorbell down"
	sleep 1
	/mnt/mmc01/mqtt_pub 192.168.178.100 1883 home/doorbell "doorbell up"
  fi
  sleep 1
done

```
[offline.sh]
```
#!/bin/sh

DEBUG_FILE=/mnt/mmc01/offline.log
#####################################################################################
# If no hosts file, leave
#####################################################################################
if [ ! -e /mnt/mmc01/hosts ]; then
  echo "error: file '/mnt/mmc01/hosts' not found" > $DEBUG_FILE
  exit 0
fi
#####################################################################################
# Wait for the actual date (comes from tuya: so no need to check if connected)
#####################################################################################
echo "Waiting for server to send date" > $DEBUG_FILE
while [ `date +%s` -lt 1645543474 ]; do
  sleep 1
done
echo "Date was set" >> $DEBUG_FILE
#####################################################################################
# Block internet access to tuya servers (change 192.168. for your private ip range)
#####################################################################################
cp /mnt/mmc01/hosts /etc
for ip in `netstat -n 2>&1 | grep ESTABLISHED | awk '{print $5}' | awk -F: '{print $1}' | grep -v '127.0.0.1\|192.168.'`; do
  echo "checking $ip" >> $DEBUG_FILE
  if [ "`route -n | grep -c $ip`" == "0" ]; then
    route add -net $ip netmask 255.255.255.255 gw 127.0.0.1
    echo "blocked $ip" >> $DEBUG_FILE
  fi
done
echo "All tuya connections blocked" >> $DEBUG_FILE
#####################################################################################
# Not needed, but ~30s faster
#####################################################################################
ifconfig wlan0 down
sleep 1
ifconfig wlan0 up
```

[cleanup.sh]
```
#!/bin/sh

#####################################################################################
# Delete streaming folders until the size is below MAX_SIZE
#####################################################################################
# - edit SERIAL for your device
# - edit MAX_SIZE to fit your SD-Card (e.g: SD-Card capacity/2)
#####################################################################################
SERIAL="xxxxxxxxx" # serial number
MAX_SIZE=16384     # max allowed folder size [MB]
MP4_PATH="/mnt/mmc01/sdrec/T-$SERIAL/mp4rec"
while true; do
  MP4_SIZE=$((`./busybox du -s $MP4_PATH | awk '{print $1}'`/1024)) # [MB]
  echo "folder size: $MP4_SIZE MB"
  if [ $((MP4_SIZE)) -lt $MAX_SIZE ]; then break; fi;
  ./busybox find $MP4_PATH/* -type d | ./busybox sort | ./busybox head -n 1 | xargs rm -r
done
```
[backup.sh]
```
#!/bin/sh

rm  backup__* 2> /dev/null
(./busybox find /mnt/mmc01/ -maxdepth 1 -type f ; echo -e "/mnt/mmc01/home/\n/mnt/mmc01/cgi-bin\n/home/"; ) | ./busybox tar -cf backup__$(date +%Y.%m.%d_%H%M).tar -T - 
# compressing is very slow on doorbell, so perhaps comment it out
./busybox bzip2 backup__$(date +%Y.%m.%d_%H%M).tar
```

guino, 2022-08-11:
@jonesMeUp nice work. The only improvement I can suggest is that I just posted a method for log_parser to work without having to kill/re-run ppsapp (which can cause problems in some devices) -- that could potentially make the startup faster, but if it is working well for you (and you don't restart often) there's really no need to change it.
Just in case, here's the information on getting the ppsapp output without killing/restarting it: https://github.com/guino/BazzDoorbell/issues/4#issuecomment-1208466146

jonesMeUp, 2022-08-12:
ah, very cool!
if there will be any ppsapp problem again, i will try your redirecting way.

it seems my ppsapp problems were:
- not killing ppsapp because the script didn't find it in /mnt/mmc01
or
- start ppsapp too soon after the kill (sleep 2 was solving the problem)

currently i have only 1 problem left:
the doorbell stream reconnect (=stuttering) depending on the running time. the longer it runs - the more it reconnects.
perhaps my sd-card is too slow, perhaps its my RTSPtoWSMP4f server?
hopefully, i will have a look at this problem next week.
currently i am using  [https://github.com/deepch/RTSPtoWSMP4f](https://github.com/deepch/RTSPtoWSMP4f) (on my pi3 with fhem/ftui) because its very easy to configure and i have the stream running in the browser (pc/android) with a few lines javascript. all tries with vlc/gplayer/etc to browser and the fu*$§%#  iframes were much more pain.

if there is a better/faster/easier open source way,  i really want to know it...

guino, 2022-08-12:
@jonesMeUp I believe there are rtsp viewer plugins you can use directly on a web page (haven't researched this in awhile), unless audio or frame rate is inportant I would just use snap/mjpeg which is what I do myself.

The connection issue you're having is likely due to crowded 2.4ghz wifi band -- there are times that it takes several seconds to get even 1 frame from my doorbell (using the phone app) because the 2.4hz wifi band here is too crowded, and if it's slow in the phone app you can be certain it's slow with everything else. Additionally my router is like 10ft from the doorbell so I know signal is not dropping.

The only thing worth mentioning is that you should be using rtsp over TCP -- the default is usually UDP and that can cause issues.

jonesMeUp, 2022-08-12:
> The connection issue you're having is likely due to crowded 2.4ghz wifi band

you are right, its 0:30 here and i had only 2 reconnects for some time. its not the online time of the doorbell, its a bad neighbours problem...

> I believe there are rtsp viewer plugins you can use directly on a web page (haven't researched this in awhile)

negative, it isn't that easy...

> unless audio or frame rate is inportant I would just use snap/mjpeg which is what I do myself.

well, framerate would be nice, audio i don't care.

> The only thing worth mentioning is that you should be using rtsp over TCP

i don't have a switch for it in my current server (i have tested it on vlc a while ago, but vlc redirecting stream was still crap - i love vlc, but only on my pc)

i am testing your redirection patch atm, looks fine at first sight. i read about strace to redirect, but there wasn't a binary for my arm processor, so i was stucked to restart ppsapp. i don't want to crosscompile right now (my brain is already overloaded with the linux stuff **YOU** forced me in with this project)

guino, 2022-08-13:
@jonesMeUp on a quick test this project worked for me for RTSP on the web browser (still requires running the RTSP client on the server): https://github.com/xpcrts/Steaming-IP-Camera-Nodejs -- this is probably similar to what you're using.

I posted the arm version or reredirect on the zip file in this post: https://github.com/guino/BazzDoorbell/issues/4#issuecomment-1208466146 it should work on all of the devices we've seen as it's compiled for armv5.

jonesMeUp, 2022-08-14:
@guino reredirect seems to work fine, i have updated the files above - hoping that my solution works now for everbody...
please add the reredirect file also to [https://github.com/guino/BazzDoorbell/tree/master/mmc](https://github.com/guino/BazzDoorbell/tree/master/mmc). 
would be nice to have all external files in 1 position.

the procejt you mentioned is ffmpeg based.
i read that ffmpeg is slow and a mess if you have multiple streams (doorbell, babycamera, etc)