guino/BazzDoorbell #83: Instructions for Merkury Smart Video Doorbell (MIC-CW014)

xraive, 2022-08-14:
I have the Merkury Smart Video Doorbell, it has firmware version 1.8.1.  Any ideas how I can get the rtsp stream for this camera?

guino, 2022-08-15:
I have never seen firmware 1.8.1 but according to this document it seems to have RTSP enabled by default with hard coded credentials: https://research.fit.edu/media/site-specific/researchfitedu/iot-lab/Geeni_Disclosures.pdf

xraive, 2022-08-15:
On the web interface I see that there are ports configured for RTSP (554) and ONVIF (36000) but I'm unable to access them.  I have root access on the device, but looking at active connections doesn't show any of the ports mentioned.  I have the following ports active on the device (8099, 8088, 6888, 23).

Using the web interface I'm able to view a stream using the url ""http://ip:8099/cgi-bin/videostream.cgi?user="user"&pwd="password"

Any ideas or suggestions would greatly be appreciated.

guino, 2022-08-16:
I would try it with VLC using the url:
`rtsp://user:password@ip:PORT` with the same user:password as the URL that works and with port being 554, 8554, 8099, 8088 to see if it works -- it may not show port 8554 active but that's the port RTSP is usually running on newer devices.
I would also try just `rtsp://ip:PORT` (no user:password) with each of the mentioned ports.

xraive, 2022-09-01:
Sorry for the delay. I tested and wasn't able to access the rtsp streams.  Thanks to you I was able to access onvif/rtsp streams on my other camera the Merkury 1080P (version 4.0.7) without any issues.  

Back to the doorbell, I did download Ghidra and using your instructions decompile the "rtspd" application and discovered 3 files that can be placed on the SD card

apexiswifi.txt - I tested and was able to successfully have the device connect the device to wi-fi using this file.
apexis_factory.txt - this one looks to connect to an FTP server.  I was able to configure it and had the device connect to FTP but I don't know how push/pull configurations.

apexis_reserved.sh - This is a script that run on startup.  I tested and was able to place command and have them executed.

I was also able to mount an nfs share as the nfs kernel module were included.

Would you be willing to take a look at the app to see what I'll need to enable onvif and rtsp?  I'm quite stuck at the moment.

Thank you for your time.

guino, 2022-09-02:
Go ahead an post a zip of the file - I will take a look when I have a chance. By your comment I assume you have the device rooted (you can login by telnet)?

xraive, 2022-09-02:
[rtspd.zip](https://github.com/guino/BazzDoorbell/files/9474383/rtspd.zip)
Yes the device is rooted.  I used the document (https://research.fit.edu/media/site-specific/researchfitedu/iot-lab/Geeni_Disclosures.pdf) to gain root access.

I've attached a zip of all the apps executed during startup.

guino, 2022-09-02:
I assume your device also has support for the TUYA app as I see a bunch of code in the file to interact with the TUYA servers ?

Assuming that's the case, please try this patched version of rtspd (kill the running rtspd in telnet and run this patched version):
[rtspd-patched.zip](https://github.com/guino/BazzDoorbell/files/9474723/rtspd-patched.zip)

If I'm looking at this right, the patched version should start the RTSP server on port 8554 (rtsp://ip:8554) hopefully without a user/password requirement (like on other tuya devices).

Let me know how that goes!

xraive, 2022-09-03:
I tried running the patched file but I'm unable to access the rtsp stream on ports (8554, 554).  I did a scan on the port themselves and they show up as closed. I wanted to mention I have the Merkury 1080P camera and the port 8554 shows up as open on my scan just for reference.

Let me know if there is anything else I can provide (config files).  Thanks for taking the time.

guino, 2022-09-03:
What phone app are you using with the device? Tuya, Smartlife, something else? I expect the device needs to be enrolled in the phone app and connected to the tuya cloud before the change I made works.

You may want to post a zip of config files and any startup scripts so I can check it out.

xraive, 2022-09-03:
I'm using the Geeni app, i'll reset the device and try it out and let you know.  I'll also upload all relevant files. 

Thanks again.

guino, 2022-09-03:
@xraive the geeni app uses the tuya cloud, so that should be ok (no need to reset and try a different app).

When you tried the patched rtspd file, were you able to use the geeni app to view the video normally? We may need to have you run the app with a redirection of the log so I can see what is happening (or you can copy/paste the output of it when you run it on telnet) — that log would be of big help.

xraive, 2022-09-05:
I've attached a copy of the partitions that include the config files and app.  I've also attached a copy of the log from the original and patched version of the app along with output of additional commands.

To answer you question I wasn't able to view video using the patched version of the app.
[rtspd.zip](https://github.com/guino/BazzDoorbell/files/9485922/rtspd.zip)

guino, 2022-09-05:
@xraive I will take a look at the files and let you know, meanwhile: can you try the same thing (kill rtspd and run it on telnet), but this time using the original/non-patched version to see if you can use the app normally? Based on the patch I made I am expecting it won’t work either and we may need to modify boot scripts to get this working.

xraive, 2022-09-05:
The original app works if I restart it from the telnet.  Unfortunately the patched version gets hung up on the section of "Waiting to connect to Wi-Fi", which is weird because the device is connected and I have telnet access already.  The check for wifi is successfull if I have the SD card inserted and i have the apexiswifi.txt file on the root of the SD card.  This doesn't happen with the original app it's able get past the "Waiting to connect to WiFi" secition successfully.

xraive, 2022-09-05:
Good news it's working.  It was my mistake.  I have rtsp working on port 8554 using the patched app.  I also have access from the Geeni app.

guino, 2022-09-05:
@xraive what did you do wrong the first time? (Just curious)

I think I saw the Tuya code waiting for connection before starting up its functions (including RTSP), so I can probably remove that wait. You will need to set the time separately (ntp server, telnet, etc) if you boot offline but it should start completely.

I don’t recall seeing anything related to onvif. I recall seeing some rtsp code unrelated to the tuya code and rtsp code for the tuya code (which is what I enabled). I will take another look but usually onvif stands out as a feature when available.

xraive, 2022-09-05:
What I was doing originally was to try running the app from /tmp directory.  I'm including my reasoning below.

I noticed when running the patched app from the SD card.  The patched app executes without support for the SD card.  This doesn't happen with the original app.  The patched app during the startup process tries to unmount and mounts the SD card.  Unfortunately, due to the patched apps location, the sd card fails to umount/mount.  The application continues to load but without SD card support.

 To resolve this issue I was trying to execute the patched app from the /tmp directory but it wasn't working for me.   My tests for the rtsp stream were when the patched app was running from the /tmp directory, which all failed.  I don't know the reason why it wasn't working but as soon as I let the patched app run from the SD card the rtsp stream started to work.  Unfortunately the patched app lacked sd card support.  Without sd card support I don't see recordings or snapshots on the sd card.

Any advice or suggestion would be greatly appreciated.  Thank you for all your efforts.

guino, 2022-09-07:
@xraive sorry for the delay. 

You can try this patched rtspd to start offline (you will need to set the time yourself with ntpd or manually):
[rtspd-patched.zip](https://github.com/guino/BazzDoorbell/files/9509230/rtspd-patched.zip)

Regarding the SD card issue during startup: You should be able to do this:
1-copy the patched rtspd to the SD card
2-telnet into the device
3-create a directory like: `mkdir /mnt/sdcard`
4-mount the SD card on the directory you created (`mount /dev/mmcblk0p1 /mnt/sdcard`)
5-kill the running rtspd process and execute the one from the SD card under /mnt/sdcard -- the rtspd process should not unmount the /mnt/sdcard directory so it should not casue problems when running the patched rtspd from that location.

Alternatively you said you could mount nfs share and you could run the patched rtspd from there. 

Have you checked if your device has writable/persistent root partition ? I have some cameras like that and it makes it very easy to modify the boot script so it runs your patched application during startup (you have a /app/bin/boot.sh file that would be perfect in that case). I also assume you can just use that apexis_reserved.sh script to kill the running rtspd and run your patched version automatically.

xraive, 2022-09-12:
@guino Apologies for the delay, I wanted to say Thank for the recent offline patch.  I did a quick test and everything is working.  Although the logs show that app is still trying to connect to the cloud. 

To clear things up a little.  I was already able to run the application from the SD card using the apexis_reserved.sh file.  The problem I have is that the when app starts it tries to unmount and mount the drive.  I'm including a partial log below.

 **### When running the patched app.**
show_version_time]: 	version time [Dec 28 2018] [16:42:48]
[show_version_time]: \*************************************************************\
[get_machine_type]: version feature is 1
[get_machine_type]: cpu type is 5
[get_machine_type]: machine type is 806
[get_machine_type]: customer is TUYAGEENI
[get_machine_type]: wifi module is 7601
[get_machine_type]: g_stMachine.customer=45
[check_packet_wifi_mode]: ret=0,wifi_mode=sta
[get_machine_type]: VERSION_TYPE is RELEASE
[get_machine_type]: smart_home is 0
[get_machine_type]: sensor type is 9
[get_machine_type]: packet name is 806.TUYAGEENI.7601.sta.NORMAL.OV2735.FV0181.1812281643
[init_product_funclist]: *********init_product_funclist 0x[0427a243] [00000000]
[init_gpio]: 
[hi_wifi_control]: wifi control status is 0
[check_safe_mode]: pin=81,enable=0
[check_safe_mode]: resetKey=0
[init_sd]: 
[umount_process]: umount /mnt/sd ret -1
[umount_process]: umount /mnt/sd again 1
[umount_process]: umount /mnt/sd ret -1
[umount_process]: umount /mnt/sd again 2
[umount_process]: umount /mnt/sd ret -1
[umount_process]: umount /mnt/sd err Device or resource busy
[mount_process]: mount /dev/mmcblk0p1 ret -1
[mount_process]: mount /dev/mmcblk0p1 again 1
[mount_process]: mount /dev/mmcblk0p1 ret -1
[mount_process]: mount /dev/mmcblk0p1 again 2
[mount_process]: mount /dev/mmcblk0p1 ret -1
[mount_process]: mount /dev/mmcblk0p1 err Device or resource busy
[get_sd_info_new]: wrong format
[init_sd]: ::SD card is exist,wrong format
[init_sd]: g_sdinsert=1
[factory_reserved]: NO SD CARD
[reboot_set_times]: reboot times is 7
[InitSysTime]: *************************InitSysTime...
[defaults_all]: 

**### **When running the original app.****
[init_sd]:
[umount_process]: umount /mnt/sd ret 0
[mount_process]: mount /dev/mmcblk0p1 ret 0
[get_sd_info_new]: mount success
[init_sd]: g_sdinsert=1
[init_sd]: ::SD card is exist
[getsdinfo]: f_blocks 7813195, f_bsize 32768
[getsdinfo]: ---total : 250019840 --- free : 249888768---
[get_sd_extra]: check /sys/class/mmc_host/mmc0
[get_sd_extra]: check /sys/class/mmc_host/mmc1
[getsdinfo]: device name is GE8QT, type is SD, serial is 9251688
[get_sdcard_info]: DeviceType:SD
[get_sdcard_info]: DeviceName:GE8QT
[get_sdcard_info]: DeviceNod:/dev/mmcblk0
[get_sdcard_info]: Partitions:0
[get_sdcard_info]: WriteProtect:0
[get_sdcard_info]: Total_partitions_space:250019840
[get_sdcard_info]: Available_partitions_space:249888768
[get_sdcard_info]: Mount_point:/mnt/sd
[get_sdcard_info]: serinumber:93016148
[unpack_cloud_library]: no cloudlib.tar.bz2
[unlink_func]: ret1=-1,/lib/libcurl.so.4,No such file or directory,errno=2
ln: /lib/libcurl.so.4: Read-only file syste`

Unfortunately, my root partition is read-only otherwise it would have been great to use the "boot.sh" script.  I tried your suggestion of running the application from an nfs share and the application is working without issue. SD card recognized this way which is great.
 
I started recording and discovered all the files have *.media extension.  Looks like i'll need their software to view it or try to convert it somehow.  Did you have a similar issue?

Thanks again

guino, 2022-09-13:
@xraive the patch does NOT prevent any cloud access -- it just **allows** it to boot completely without internet. If you don't want it to use internet you have to block access on your router or using hosts file (see: https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Use-the-device-OFFLINE-(disconnected-from-tuya-servers)-%3F )

I'm not sure what the .media format is -- the ones I have seen are .data files -- you either need to get a converter from the manufacturer OR save them using the phone app (so they're converted) OR you may be able to play them without audio using VLC if you force H264 demuxer, or at least that's how it is with .data files I have (and I have no converter for them).

xraive, 2022-09-13:
Hi Guino, yes that's how I have it setup right now.  I'm blocking the device on my firewall.  Although I have to first initialize the device using the app so that settings like Wi-Fi are persistent after reboots.

In regards to VLC I'll try to see if I can open the .media file.  As another option I have MotionEye accessing the camera using rtsp, I've done some test and I am able to record that away in the format that I want.  It will help reduce the load on the camera.

Question  for you and I don't know if this is possible is there a way to increase the number of streams available?  Currently I can only have 1 RTSP stream.

guino, 2022-09-14:
@xraive devices with ONVIF code usually have 1 low resolution stream and 1 high resolution stream (/SteeamingChannels/101 and 102). Devices with just RTSP only have 1 stream (matching resolution of the camera.

Shaztech, 2023-09-07:
I have the same camera (doorbell), I can telnet with the same credential. I've tried to run the rtspd-patched file on the card. It run fine but the rtsp ports are still blocked. Also, the default mount point for the sd is /mnt/sd and I cannot create any directory in /mnt. Can you help ?