guino/BazzDoorbell #84: Can this be flashed? Bell 7S ppstrong-b5-tuya_general-3.3.0.20210816

leonardochaia, 2022-08-27:
Hi all.

I think my doorbell cannot be flashed but just wanted to make sure before I find another solution or give it away

I added the `ppsFactoryTool.txt` and got access to port `8090`

`/devis/deviceinfo` does respond:
```
 curl http://admin:056565099@192.168.55.131:8090/devices/deviceinfo                                                                              INT ✘  2m 7s  
{
	"devname":	"Smart Home Bell",
	"model":	"Bell 7S",
	"serialno":	"058627181",
	"softwareversion":	"3.3.0",
	"hardwareversion":	"",
	"firmwareversion":	"ppstrong-b5-tuya_general-3.3.0.20210816",
	"licence_id":	"<redacted>",
	"licence_key":	"<redacted>",
	"authkey":	"<redacted>",
	"deviceid":	"<redacted>",
	"pid":	"vFHpaEFwu8UDZnAI",
	"WiFi MAC":	"ac:5d<redacted>",
	"ETH MAC":	"08:88<redacted>",
	"mcuversion":	"4.4.3.20210318"
}
```

However, `/proc/cmdline` hangs.
From what I figured in other issues I've read this is because it's not running linux.

Would someone confirm it please?

Thanks, and thanks for all the effort put on this repo!! Amazing work.

Leo.

guino, 2022-08-27:
@leonardochaia I have never seen that firmware version. I have seen a 3.1 firmware before and it was a non-linux firmware. I have never seen a ‘battery operated’ device run linux as it would just drain the battery away, so if your device is battery powered (or has a battery only power option) it likely runs non-linux firmware (ie RTOS).

leonardochaia, 2022-08-28:
Understood. Yeahp, it' has a battery power option.
Appreciate the help and quick reply!

Guess I'm gonna build it myself.

Just in casse for anyone looking after this, the brand was "Linda Smart" (Argentina)
Linda Smart Doorbell Tuya Compatible