guino/BazzDoorbell #85: ppsapp patch request for LSC smart video doorbell 5.2.2 (ppstrong-a3-tuya2_general-5.2.2.20210914)

renzenicolai, 2022-09-09:
Hello, I'd like some help with patching the ppsapp for my device. I tried but can't figure out how to enable RTSP  / onvif via browing the decompiled code with Ghidra. Could you help me (and others with this model / version) by patching the ppsapp?

Model: Bell 8S
Hardware version: BE8S_A2_V10_433
[ppsapp.zip](https://github.com/guino/BazzDoorbell/files/9537300/ppsapp.zip)

Software version: 5.2.2
Firmware version: ppstrong-a3-tuya2_general-5.2.2.20210914

ppsapp hash: a408d7c0373421cff8cfa8aa755672cd

I've attached the original ppsapp.

guino, 2022-09-12:
@renzenicolai Sorry I am traveling and haven´t had a chance to look at it just yet -- I will see what I can do as soon as I can.

guino, 2022-09-13:
@renzenicolai your ppsapp does not have the standalone RTSP function/code like 2.7.x and 2.9.x firmware has. Your only option is to enable onvif using tuya_config.json -- you do have to register the device in the tuya/manufacturer app before you can view/edit tuya_config.json. Basically:
1-telnet into device
2-execute `cp /home/cfg/tuya_config.json /mnt/mmc01`
3-wait 5 seconds, eject SD card, edit tuya_config.json file
4-execute `cp /mnt/mmc01/tuya_config.json /home/cfg`
5-wait 5 seconds, reboot device

johan-van-marion, 2023-02-27:
I've got a Nedis smart video doorbell with the same firmware:
5.2.2 (in the Tuya App)
Tried to telnet into the device but port 23 isn't open ...
Could not open connection to the host, on port 23: Connect failed

Any help is appriciated, need to configure the rtsp for integrating into HA.

guino, 2023-02-27:
@johan-van-marion that's not accurate. The instructions say to follow the steps from https://github.com/guino/Merkury720 (basically ten steps with a lot of information), but the 3 files copied over the original ones (listed in the ten steps I mentioned). Step 4 will have you boot with the reset button pressed to actually install the hack. Once installed you should have telnet access which will allow you to modify your tuya_config.json file.

johan-van-marion, 2023-02-28:
@guino Thanks for that missing bit (reading patiently is not my strong suit) but still some questions, before i boot the device.
Can i replace the 3 files directly on the SD card, or do I  need to follow the 10 steps and re-do it with the modified files again ? 
I now have this as content of the SD-card:
[SD-Card.zip](https://github.com/guino/BazzDoorbell/files/10847958/SD-Card.zip)

I only need to enable the Onvif to link it to my HA.
And do i need to keep the SD-card in even after Onvif is enabled?

guino, 2023-02-28:
@johan-van-marion your SD card files seem correct - you should be able to continue from step 4 of https://github.com/guino/Merkury720 

Once ONVIF is enabled on your device (tuya_config.json) you should not need the SD card in the device unless you want to use telnet, download files, or integrate it somehow (i.e. motion notifications).

johan-van-marion, 2023-02-28:
@guino Did everything from step 4: 
even did the modification in "dev" and changed 30 to 60
But still nothing happends.
non of the below addresses are reachable in "chrome" 
http://admin:056565099@<ip>:8090/proc/cmdline
http://admin:admin@<ip>:8090/proc/cmdline
http://admin:056565099@<ip>/proc/cmdline](http://admin:056565099@ip/proc/cmdline)
http://admin:admin@<ip>/proc/cmdline](http://admin:056565099@ip/proc/cmdline)
http://admin:056565099@<ip>:8090/devices/deviceinfo 
http://admin:admin@<ip>:8090/devices/deviceinfo
.....
or any other url's described.
Am i missing something ? 
Did insert the sd-card and hold the reset button for 5 seconds and waited sevral minutes....

guino, 2023-03-01:
@johan-van-marion looking at your SD card it seems files (posted above) it seems you missed the step to create ppsFactoryTool.txt file. Without that file none of the URLs will work (you have to boot the device with this file in the SD card for it to work). I also wanted to make sure:you are putting the IP address on the URLs -- for example, if the camera IP is 192.168.1.123 you're trying: http://admin:056565099@192.168.1.123:8090/proc/cmdline (and also with admin:admin). if ppsFactoryTool.txt is created and the URLs are not working then your camera may just not be compatible with the ones we've seen so far.

johan-van-marion, 2023-03-01:
http://admin:admin@192.168.68.158:8090/proc/mounts

rootfs / rootfs rw,size=17160k,nr_inodes=4290 0 0
proc /proc proc rw,relatime 0 0
tmpfs /tmp tmpfs rw,relatime 0 0
var /var tmpfs rw,relatime 0 0
sysfs /sys sysfs rw,relatime 0 0
devpts /dev/pts devpts rw,relatime,mode=600,ptmxmode=000 0 0
/dev/mtdblock2 /opt/pps squashfs ro,relatime 0 0
/dev/mtdblock3 /home/cfg jffs2 rw,relatime 0 0
/dev/mmcblk0p1 /mnt/mmc01 vfat rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,usefree,errors=continue 0 0

johan-van-marion, 2023-03-01:
http://admin:admin@192.168.68.158:8090/proc/self/root/home/cfg/tuya_config.json

{
"version":1,
"sleep_mode":0,
"alarm_fun_onoff":0,
"alarm_fun_sensitivity":0,
"alarm_fun_mode_switch":0,
"alarm_fun_time_start":0,
"alarm_fun_time_end":0,
"flip_onoff":0,
"light_onoff":0,
"night_mode":0,
"sound_detect_onoff":0,
"sound_detect_sensitivity":0,
"watermark_onoff":1,
"event_record_time":60,
"enable_event_record":1,
"record_enable":1,
"motion_trace":1,
"motion_area_switch":0,
"motion_area":"",
"motion_tracking":0,
"cry_detection_switch":0,
"humanoid_filter":0,
"loudspeaker_vol_pct":100,
"jingle_mode":1,"jingle_sound":0,
"jingle_volume":100,
"jingle_exist":0,
"flight_main_mode":0,
"onvif_enable":0,
"onvif_pwd":"admin",
"pan_default":-1,
"tilt_default":-1
}

So the .json file is on my device now a way to pulling it off 👍

johan-van-marion, 2023-03-01:
@guino Is a 2GB SDcard big enough for doing this?

johan-van-marion, 2023-03-01:
> @guino Is a 2GB SDcard big enough for doing this?

O.k. just checked with and other card (128GB) but still same HTTP 500 error

guino, 2023-03-01:
some people have issues with the /hack URL even when it works -- check the SD card to see if the 'home' directory was created during boot, if so, the hack worked.

Another way of know if the hack is installed is to simply check if the /proc/cmdline URL changed.

Unfortunately some devices are very picky about the SD cards to install the hack (the SD card may work after the device starts up but doesn't work to install the hack) so if your /proc/cmdline URL didn't change you may need to install it with try different brand SD cards (step 4 where you push the reset button on power on).

johan-van-marion, 2023-03-01:
@guino, so far i tried 4 different sd-card:
1 GB Sandisk
2 GB Transcend
32GB Lexar
128GB Samsung
All the same result
Also tried different busybox (saw there is some differend out there) all in all NO-GO so far.
cmdline always seems to be the same
Which Busybox should be the right one? Could you send it here?
I feel i'm close but still missing something....

johan-van-marion, 2023-03-01:
Here's some out put of the "hack" 
it dosn't help me,b ut maybe you see what can be wrong:
curl "http://admin:admin@192.168.68.158:8090/proc/self/root/mnt/mmc01/hack" -v
*   Trying 192.168.68.158:8090...
* Connected to 192.168.68.158 (192.168.68.158) port 8090 (#0)
* Server auth using Basic with user 'admin'
> GET /proc/self/root/mnt/mmc01/hack HTTP/1.1
> Host: 192.168.68.158:8090
> Authorization: Basic YWRtaW46YWRtaW4=
> User-Agent: curl/7.83.1
> Accept: */*
>
* Mark bundle as not supporting multiuse
< HTTP/1.1 500 Internal Server Error
< Content-Length: 0
<
* Connection #0 to host 192.168.68.158 left intact

guino, 2023-03-01:
@johan-van-marion is your cmdline still showing this:
```
console=devnull mtdparts=spi0.0256K@0x0(BOOT),3072K@0x40000(sys),4288K@0x340000(app),448K@0x770000(cfg),64K@0x7E0000(enc),64K@0x7F0000(sysflg) mem=64M memsize=64M pcbversion=BE5S_A5_V10 sensor=gc2063mipi model_name=Bell-5S
```

If so the issue is in applying the hack (busybox won't change anything). In any case this is the right busybox: 
https://github.com/guino/Merkury720/blob/main/mmc/busybox?raw=true (about 1Mb in size)

johan-van-marion, 2023-03-01:
@guino so nothing seems to be wrong on the sd-card any more, it's just the device which is the problem? 
Anyway i attached the final  version of the SD-Card. 
I think everything i've done is according to steps 1..6
Note: edditted the : ppsFactoryTool.txt and removed my SSID and PASSWORD
[SD-card.zip](https://github.com/guino/BazzDoorbell/files/10865390/SD-card.zip)

johan-van-marion, 2023-03-02:
@guino, just checking #2 but could it be the reason it's not working due to the fact i didn't modified the http.conf file?
it's still having : /:user:password
and should i change it to : /:admin:admin
Is that why i'm getting the HTTP 500 error?

honarkhah, 2023-03-02:
Hi, I have tried the flow, but the camera does not recognize the file ppsFactoryTool.txt and it is not able to join the wifi network, I have attached the device detail.
LSC smart indoor camera: version: 7.6.32
<div style="">
[image]
[image]
[image]
</div>

guino, 2023-03-02:
@johan-van-marion like I said, we need to focus on trying to get your /proc/cmdline to change -- for that to work the only files required in the SD card are: env, ppsMmcTool.txt, ppsFactoryTool.txt

Verifying that these 3 files are in the card, power on the device with the SD card already inserted then see if you get a response to any of these URLS:
http://admin:056565099@192.168.68.158:8090/proc/cmdline
http://admin:admin@192.168.68.158:8090/proc/cmdline
http://admin:056565099@192.168.68.158/proc/cmdline
http://admin:admin@192.168.68.158/proc/cmdline

If your /proc/cmdline response (URLs above) doesn't show `- ip=30;/mnt/mmc01/initrun.sh)&:::::;date>/tmp/hack;(sleep` then none of the other files in the SD card will have any effect.

Additionally, alternatively to getting the URL http://admin:admin@192.168.68.158:8090/proc/self/root/mnt/mmc01/hack to work you can simply check if there's a file named 'hack' on the SD card -- this file is created if the hack was successfully installed. Again, without that `ip=30` section in the /proc/cmdline response this will never happen.

I have seen some rare cases in which the device had a defective 'reset button' and the user in question exchanged the device under warranty and the new device worked correctly. I'm not saying this is your case, but it is a possibility.

If you can't get the /proc/cmdline to show the `ip=30` section then your only option would be to open the device, remove the flash chip, modify the tuya_config.json file in the chip and solder it back. 

Once/if you have correct /proc/cmdline response we can then check httpd.conf and such but that's definitely not related to the 500 error you're seeing.

johan-van-marion, 2023-03-02:
Tried all the above, so i think i'm stuck until some one else finds a solution