guino/BazzDoorbell #86: Questions

menn0, 2022-09-22:
Dear Guino,

I have the ieGeek version, two questions related to this doorbell;

1. In the Cloudedge app we can switch between 3 streams, SD, HD and FHD, in RSTP there are only two channels available 101 and 102. Why is this? are there in fact two or three streams?
2.  What is the FPS of this camera, I can only get 10fps out of it

guino, 2022-09-22:
@menn0 I'm unfamiliar with your camera, but all TUYA code I have seen only deals with 2 streams (SD and HD) the HD stream is usually whatever the resolution of the camera (720p or 1080p). I have not seen any code handling 3 streams so your camera or phone app may be doing some conversion of the stream and/or quality.

These devices usually have low fps due to the low powered cpu in them. My recordings 'report' 30 or 15fps but I think actual fps it's lower than 15.

AngelChezziPXL, 2023-02-10:
@guino Hello Guino
Some months ago I bought a LSC smart video doorbell at an Actions store in Belgium. I would like to flash it so that I can use it offline without the TUYA cloud on HA. I followed all the steps you wrote on the Merkury 720P and the Doorbell repos, but untill now I did not succeed to flash the firmware. 
I do not have big knowledge of uboot. Could you please help me to clone the original firmware to a SD card so that I can see what is in it?
Thanks.
Kind regards

guino, 2023-02-10:
@AngelChezziPXL you could first try the steps described here: https://github.com/guino/BazzDoorbell/issues/11 -- if that doesn't work then you'd have to be comfortable with opening the device and either using a hardware programmer to read the flash OR trying to use the serial port (both options will likely require soldering work). Unfortunately there's not much I can do without the device itself.

If you were able to get any information from the device with /devices/deviceinfo or from the phone app (i.e. firmware version, model, picture) that could tell us if there's a specific process that may work for it (possibly without opening the device). If your device is battery operated there's little chance that any of the mods would work for it.

AngelChezziPXL, 2023-02-13:
Hey Guino
Thanks for the fast answer. I'll try it out and keep you posted to let you know if it worked.

AngelChezziPXL, 2023-02-13:
Hello Guino
I just tried to copy the memory chip following the steps instructed in #11 , but the bin file was not created on the SD card. I tried both versions: mini7c.zip and ppsMmcTool.txt. I also adapted the address in the ppsMmcTool.txt and env file to my address (81c0800). But it still doesn't work. 
Here below you find a snapshot of what I read with the UART (I added some red markings next to the errors):
[image]
[image]

Could it be because I tried al the steps in of the 'Merkury720p'? 
When the IPcam was 'virgin' I could stop the uboot, but since I did the first attempt to flash it it counts down 3 second, shows two times 'magic err' and continues booting.
How can I reset everything to standard to try it again?

guino, 2023-02-15:
@AngelChezziPXL the steps in #11 don't create a file in the SD card -- they write the data to the SD card directly (not in a file) so you have to use 'dd' (on a computer) to create the file on the computer (Step 13). It should always create a file even if the process didn't work (then we can check if the file actually has a copy of the firmware).

The 'magic err' message always shows up - you can ignore it.

From your UART output it looks like the data was written to the SD card, so if you use step 13 you should get a copy of the flash.

Since you have UART access feel free to send me an email (see my github profile) and I can give you some info to try things on the UART.

There's no need to 'reset everything' -- #11 doesn't modify the device in any way. If you have ran the other processes then booting without the SD card should allow the device to boot normally (without changes).

AngelChezziPXL, 2023-02-15:
@guino  A very big thanks for the explanantion. I tried it again and now it worked. I've created an flash.bin containing:
- uImage header
- cramsFS filesystem
- two jffs files
- multiple Zlib files
- two xz files
- multiple ELF
- multiple Unix path

What I've to do now so that I can adapt it to have the IP doorbell work in Home Assistant?

Thanks
Kind regards

guino, 2023-02-16:
@AngelChezziPXL can you provide a copy of the flash.bin ? (i.e. you can email it to me). I'd like to check why the rooting process didn't work on it -- I'm assuming you already tried the steps here: https://github.com/guino/Merkury1080P#conclusion (based on the UART log that is the process that has the best chance to work for your device).

AngelChezziPXL, 2023-02-16:
@ guino Ok I will send you the file.

I will try the steps(conclusion) again. For the moment when I netcad the ip address of the video doorbell I only see 1 port open. No webpages or other things are visible. The only possible things is to video stream of the camera to VLC.

Do you know how I can set again that the bootloader counts down 5 second before booting? Now it counts down 3 second, but there is no way to interrupts the booting. I think it has been disabled when trying to flash it, beacause when it was new it showed a message like 'hit any key to stop the boot' and it counted down 5 seconds. When pressing a key it asked for a password in the past, now it doesn't do it anymore

guino, 2023-02-16:
@AngelChezziPXL Based on the provided flash file it looks like that device should work with the steps I provided and in fact it has already been reviewed/reported as working: 

ppstrong-a3-tuya2_general-4.0.7.20210513 | BE8S_A2_V10_433 | 96729b630ee4f088fda3307a2370d050 | Bell 8S
-- | -- | -- | --

<p dir="auto">This firmware does NOT need patching to use ONVIF/RTSP -- just enable it as described <a href="https://github.com/guino/Merkury1080P/issues/9#issuecomment-926414826" data-hovercard-type="issue" data-hovercard-url="/guino/Merkury1080P/issues/9/hovercard">here</a><br>
snap.cgi and mjpeg.cgi address: 0x0395cbc<br>
play.cgi request address: 0x03964cc

If you want to make it work 'offline' we need to make an offline patch so the device can start up completely without connecting to the tuya servers (or you can try the universal offline method): https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Use-the-device-OFFLINE-(disconnected-from-tuya-servers)-%3F

AngelChezziPXL, 2023-02-16:
@guino Hello Guino
Actually I already did the steps. The problem is that I only receive the ONVIF/RTSP stream, but the 2 way communication is not working and I cannot trigger when the button of the doorbell is pressed.

If I take the option 1 of the link you sent me for the offline usage, do you really need internet acces each time the bell powers up again, or just the first time when you set it up?
Does the 2 way communication work and is there a message for the button press?

I've also another question. How can I enable telnet and reactivate that a key pressed during boot-up interrupts the boot process?
Thanks

guino, 2023-02-16:
@AngelChezziPXL There's no support for 2 way communication since we only get the RTSP/ONVIF feed. At one point I saw someone making a custom application with NodeRED to record/transmit playback on the device, but that likely wouldn't work with HA anyway.

Offline mode Option 1 does require the internet to be available every time the device powers up.

For telnet:
-make sure you have the correct busybox in the SD card, you can download it from this link: https://github.com/guino/Merkury720/blob/main/mmc/busybox?raw=true (over 1MB in size)
-make sure you followed step 9 from this link: https://github.com/guino/Merkury720 (creating a passwd file or modifying custom.sh)
-make sure custom.sh has these lines (for password protected telnet):
```
cp /mnt/mmc01/passwd /etc/passwd
/mnt/mmc01/busybox telnetd
```
OR (if you want passwordless telnet) this line:
```
/mnt/mmc01/busybox telnetd -l /bin/sh
```

The button press notification is something you can do with log_parser.sh, for your device this is the best method: https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Use-notifications-MQTT-log_parser-without-a-patched-ppsapp-%3F

You will need to edit the notification commands inside log_parser.sh (one for doorbell notification and another for motion detection if you want) -- this will include IP, subject, message information as configured in HA. 

I recommend having telnet working so you can execute the notification commands manually to simulate the button press so you can check that HA is receiving it. I only use log_parser.sh with homebridge, so I'm not familiar with how HA should be configured to receive MQTT notifications (you may also be able to do HTTP notifications). Here's some info on that if you haven't seen it yet: https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Integrate-with-Home-Assistant,-HomeBridge,-Domoticz,-etc

AngelChezziPXL, 2023-02-16:
@guino 
Thanks for the clear answer. I'll give it a try during the weekend. Did you get it work in HA?

I'm impressed of your knowledge. How did you figure it all out? ;-)

Ah ... yes, I red somwhere on your git that you like beer. Do you like Belgian beer and chocolats?

guino, 2023-02-17:
@AngelChezziPXL sorry, I don't have and don't use HA -- I use Domoticz + Hoobs for home automation control. I tried HA one (long time ago) and it was too big (disk), too heavy (RAM) and too slow (CPU) -- granted I was using a raspberry pi.

Been coding for 30 years, so I learned a lot along the way.

Who doesn't like beer and chocolate ? I'm not picky about origin... :)

AngelChezziPXL, 2023-02-19:
@guino Is it possible to have 2-way communications in Domotics with my doorbell?

I'm a beginner :-)

I'll check if it is autorized to send beer by post to Canada (I supose you live in Canada).

AngelChezziPXL, 2023-02-19:
@guino 

> @AngelChezziPXL There's no support for 2 way communication since we only get the RTSP/ONVIF feed. At one point I saw someone making a custom application with NodeRED to record/transmit playback on [...]

@guino It worked! Thanks.
Telnet is now working. I noticed that doorbell chip heats up a lot. Is it normal?

guino, 2023-02-20:
Some devices heat up more than others, I don't have any device like yours so I can't compare. 
I don't think the one I have heats up much, but it could be related to the voltage you're running on your doorbell (mine runs on 8V).