guino/BazzDoorbell #87: Corrupted rtsp stream

jurasjo, 2022-10-10:
Hi,
I wonder if anybody has similar problem with his Tuya based WIFI camera. I have 3 similar cameras (NEDIS SmartLife Outdoor Camera) - [https://nedis.com/en-us/product/security-safety/ip-security/outdoor/550671323/smartlife-outdoor-camera-wi-fi-hd-720p-ip65-internal-16gb-12-v-dc-night-vision-android-ios-silver-white?_ga=undefined](url)
I'm attaching the link to the screenshot: [https://prnt.sc/qdfOiXpHIOuW](url)

The problem is related only to the main rtsp stream (1920x1080). The rtsp substream always works. Video in Nedis App works normally but sometimes it has a problem to load and at the same time rtsp doesn't work as well. Issue started while camera were on "stock" firmware (2.10.6) and it is also the case now while hacked.

Somebody any idea what can be wrong?

guino, 2022-10-10:
@jurasjo never seen that issue myself, but it seems like it could be due to wifi issues (bandwidth/reliability). I would make sure you're using RTSP over TCP also, but otherwise I have no other suggestions. Does the problem stay for awhile ? if so, does rebooting the device fix it ?

guino, 2022-10-10:
@jurasjo while the image is bad (as on your screenshot) -- if you open the phone app, does it look the same or does it look correct ? If it looks the same then it's hardware/firmware issue, if it looks fine it's an issue with RTSP, Wifi or client.

Is the firmware 5.2.1 not available for upgrade on your older cameras ?

jurasjo, 2022-10-10:
This kind of crappy stream occurs only while using RTSP (VLC, Frigate, Home assistant) and only on main stream. Sub stream always works. Image on the phone is correct. I would suspect some incompatibility with WIFI module connected to firmware 2.10.6.

There is no upgrade available. If I remember correctly those 3 cameras had the same base firmware and then there was an upgrade to  2.10.6 and the fourth one had different base firmware(I think 5.x.x) and went to 5.2.1.

jurasjo, 2022-10-11:
Hi I use rtsp://admin:admin@192.168.13.60:8554/Streaming/Channels/101 and /102.  At this very moment main stream doesnt'work, substream works and both HD and SD steams work in Mobile app.

I have attached ppsapp (from /home/app? if I'm recognising correctly).
[ppsapp.tar.gz](https://github.com/guino/BazzDoorbell/files/9757697/ppsapp.tar.gz)

jurasjo, 2022-10-11:
And now cameras went to night mode(black-white) and all the streams works in HD ;)

jurasjo, 2022-10-22:
Have you had a chance to look into this file?

guino, 2022-10-24:
@jurasjo sorry, I was traveling for the whole last week -- I'll take a look as soon as I have a chance.

jurasjo, 2022-10-24:
Not a problem. I'm wondering what had changed that made it stop working(HD stream). All of those cameras at once. It worked on this firmware (2.10.6) for at least half.

guino, 2022-10-29:
Try this patch with rtsp://IP:8555 - see if that RTSP version works better for you.
firmware: ppstrong-c51-tuya2_teco2-2.10.6.20210814
md5: 9738e84be3de15bc99b65dcdf797946c
snap/mjpeg.cgi address: 0x056e6b0
play.cgi address: 0x0571f94
offline+rtsp port 8555 patch: 
[ppsapp-rtsp.zip](https://github.com/guino/BazzDoorbell/files/9894055/ppsapp-rtsp.zip)

jurasjo, 2022-10-30:
Thank you for your help and patience ;).
So, I assume I have to follow this procedure:
`Here's how to patch ppsapp
    After the hack is installed: power off the device, remove the SD card and take it to a computer
    Go to https://www.marcrobledo.com/RomPatcher.js/ DO NOT CLICK ON CREATOR MODE
    Click ‘choose file’ in front of ‘ROM file’ and select the original ppsapp file under /home/app/ppsapp on the SD card (assuming you didn’t do anything to it since the hack was installed)
    VERIFY that the md5 value displayed matches the md5 from the list below (listed with the patch for your ppsapp)
    Click ‘choose file’ in front of ‘Patch file’ and select the zip file you downloaded from the list below
    Click ‘apply patch’ and save/download the file to the root of the SD card (the default file name will likely be ppsapp-rtsp)
    Rename the saved/downloaded file on the root of the SD card to ppsapp (make sure it has no .txt or any extension) — verify the size is exactly the same as the size of /home/app/ppsapp in the SD card (which should be a few megabytes)
    Properly eject/unmount the SD card from computer (i.e. windows using the tray icons, linux umount command, etc)
    Insert SD card to device and power it on
    Wait for it to boot - you likely will hear the startup sound 2 times. After the 2nd startup time you can try to view the rtsp feed under rtsp://ip:8554 (unless you're on 2.9.7 in which case you use the links provided with the patch).
`
I have done also some online searching about the possible problem with Omada equipment and there may be some issue regarding WIFI settings. I have changed some settings and for some time it works as it should but it may be coincidence.  I will set up my WIFI as it was previously, then I will wait for the stream to be corrupted and then I will apply fix you provided. 

I will report within few days.

guino, 2022-10-30:
@jurasjo Yes -- the only thing to note is that your rtsp URL will be rtsp://IP:855**5** (NOT 855**4**)