guino/BazzDoorbell #88: w2 doorbell not WiFi access point

sacredx72, 2022-10-21:
Good afternoon, I became the owner of Tuya W2 door glass
Tuya APP shows main module version V3.5.17, MCU version 3.5.17
https://www.aliexpress.com/item/1005003443413635.html
When I try to place the ppsFactoryTool.txt file on the SD card, the device boots up, resets all settings (operation mode, motion sensor settings), but remains connected to my main WiFi network and in the Tuya application.
While the file is on the SD card, all subsequent restarts of the device also reset the settings.
But at the same time port 80 and 8080 does not open.
With the ppsFactoryTool.txt file marked up, a logfile.txt file is created on the map, and judging by the contents in my linux device =) 
```
`ЭМ»Є   –    @        яяяяЂT Ђd ЂT ЂT ЂT ЂT pU$T 	\w(	\w@	\w¤	\w                                    ‰                                                                                                                                              <<<<<powerup=0>>>>
                    [     1671][ULOG_INFO]main[4155] ###Ver:1.0.7.70
[     1763][ULOG_INFO]ubiaGetVd[334] read UBIA_VD_START=90000,len=10000
[     1764][ULOG_INFO]ubia_get_local_mgmt[4288] local gUID:FICE2UFBS3OJASF7IE7A PKG:117 ModelNum:10!
[     1766][ULOG_INFO]ubia_load_user_conf[3705] get dns0:01a9a8c0 dns1:00000000
[     1766][ULOG_INFO]ubia_load_user_conf[3706] u8RecordMode=1 country=2, wifistatus=0
[     1766][ULOG_INFO]ubia_load_user_conf[3771] ###############AP:      len:0,u8HostApPreFix=TUYA
[     1766][ULOG_INFO]ubia_load_user_conf[3787] ###############AP:TUYA_FICE      len:9
[     1766][ULOG_ALERT]is_invalid_kuid[3490] is_invalid_kuid str=0x11572cd,len=12
[     1766][ULOG_ALERT]Ubia_GetUpdateCrc[1162] open file:/config/UpdateCrc.ini error
[     1766][ULOG_INFO]Ubia_GetUpdateCrc[1229] crcType 2,crc=0

[     1829][ULOG_INFO]ubia_wow_loadserver[92] err magic: 00000000
[     1845][ULOG_INFO]mqueue_open[88] bind sim_socket [3] to port:88
[     1845][ULOG_INFO]main[4247] set hasValidUID=1
[     1845][ULOG_INFO]IPC_APP_Init_SDK[147] s_mgr_info.dev_sw_version:3.5.17
[     1845][ULOG_INFO]IPC_APP_Init_SDK[154] IPC_APP_Init_SDK, product_key[eitdf2gzpcrmq9rm]
[     1845][ULOG_INFO]IPC_APP_Init_SDK[157] Init Value.uuid hwxd8e2c9fa924a6336d
[     1845][ULOG_INFO]IPC_APP_Init_SDK[158] Init Value.auth_key TmvZJaXF5vyA6WxiwVL9FlzAdNN2IbDd
[     2051][ULOG_INFO]hwl_wf_set_country_code[893] Set Country Code:CN 
[     2052][ULOG_INFO]IPC_APP_Init_SDK[184]  tuya_ipc_start_sdk success 
[     2052][ULOG_INFO]main[4266] tuya_audio_ample:8000
[     2053][ULOG_INFO]hi_3861_init[202] hi_channel_init, ret:0
[     2053][ULOG_INFO]hi_3861_init[204] hi_channel_register_rx_cb, ret:0
[     2053][ULOG_INFO]get_wifi3861_status[672] wifi mode:0,dev_type=0,wifi_country_code:0,0,netcheck_interval=0,log_enable:0
[     2053][ULOG_INFO]get_wifi3861_status[673] protect_enable:0,60,3
[     2053][ULOG_INFO]get_wifi3861_status[675] ###qeust 3861 status###
[     2053][ULOG_INFO]hwl_wf_wk_mode_set[659] hwl_wf_wk_mode_set mode:2
[     2055][ULOG_INFO]get_wifi3861_status[679] hi channel send MSG_WIFI_HI3861_STATUS_REQ success,len:164
[     2055][ULOG_INFO]ubia_network_start[2165] init netstatus=0
[     2060][ULOG_INFO]ubia_secondwork_mgmt[646] ####secondwork_mgmt
[     2098][ULOG_INFO]main[4333] (u8WorkMode:0)
[     2099][ULOG_INFO]ubia_get_video_stream[683] ubia_get_video_stream ubia_t31 run
[     2180][ULOG_INFO]ubia_parse_hi3861_data[1836] MSG_WIFI_HI3861_STATUS_RSP 2180
[     2180][ULOG_INFO]ubia_set_time[1671] set data:date -s "1970-01-01 00:00:02"
[     2183][ULOG_INFO]USART_Send_Data[291] mcu send=7B,04,EE,6E,
[     2224][ULOG_INFO]ubia_parse_hi3861_data[1847] tuya_ipc_set_service_utctime:2
[     2224][ULOG_INFO]ubia_parse_hi3861_data[1854] iswakeup:0,wakereason=0,wifi_ver=0x0206
[     2225][ULOG_INFO]set_wifi3861_plugout[853] hi channel send MSG_WIFI_HI3861_MODE_REQ,mode:9 success,len:40
[     2225][ULOG_INFO]ubia_parse_hi3861_data[1985] wifi ver:0x0206, p_ubia_hal_mgmt->wifi_dataver:0x3f000206
[     2225][ULOG_INFO]ubia_parse_hi3861_data[1986] wifi rssi:0, get hi3861 current time:2, pirlevel:0
[     2225][ULOG_INFO]ubia_parse_hi3861_data[2013] 22 power_value:3837,index=299,powerSupplying=0,resv=0
[     2225][ULOG_INFO]ubia_parse_hi3861_data[2015] iswakeup:0, wakereason:0,u8WorkMode=0
[     2225][ULOG_INFO]ubia_parse_hi3861_data[2053] UNET_INIT
[     2225][ULOG_INFO]ubia_check_pairOrSendConnectInfo[1906] p_ubia_net_mgmt->netstatus:0
[     2225][ULOG_INFO]ubia_check_pairOrSendConnectInfo[1914] #### connect wifi########
[     2225][ULOG_INFO]ubia_sta_connect_wifi[29] wifi ssid:Wi-Fi Sacred Home,psw:sacred007,auth:3!
[     2225][ULOG_INFO]ubia_sta_connect_wifi[34] #######send ssid key to  hi3861 ########
[     2225][ULOG_INFO]config_wifi3861_net[1553] send wifi uid:FICE2UFBS3OJASF7IE7A,req.zoneid=2
[     2226][ULOG_INFO]config_wifi3861_net[1557] hi channel send MSG_WIFI_HI3861_CONNET_REQ success,len:160
[     2226][ULOG_INFO]ubia_init_eventtype[2558] iswakeup=0,u8WorkMode=0,wifistatus=0,wakereason=0,isplugout=0
[     2226][ULOG_INFO]ubia_init_eventtype[2622] eventType=1
[     2226][ULOG_INFO]init_wifi3861_pirmode[942] hi channel send 3861 pirmode :1 success,len:40,operation=5
[     2227][ULOG_INFO]ubia_parse_hi3861_data[2139] wifi rssi:127
[     2227][ULOG_INFO]ubia_parse_hi3861_data[2140] power_mode:0,adc_value:3837,index:299,battery:100
[     2250][ULOG_INFO]get_reset_value[2535] open gpio46 success
[     2253][ULOG_INFO]get_video_stream2[441] tuya_time_sync ok,start read video
[     2253][ULOG_INFO]get_video_stream2[522] first frame video timestamp=451,pts=0
[     2267][ULOG_INFO]USART_Receive_Data[358] mcu rec=7B,0A,9D,C1,00,0D,6A,01,2D,99,
[     2267][ULOG_INFO]Uart_CMD_Handle[881] READY RESP:c1 00 0d 6a mcu[01 2d]
[     2280][ULOG_INFO]ubia_light_OnOff[891] led status: off
[     2464][ULOG_INFO]init_fdk_decoder[111] transportFmt:0
[     2465][ULOG_INFO]init_fdk_decoder[112] conf_len = 4`
```
Maybe take a look at the log file and help in trying to enable the RTSP stream

sacredx72, 2022-10-21:
[logfile.txt](https://github.com/guino/BazzDoorbell/files/9839622/logfile.txt)

sacredx72, 2022-10-21:
Board

guino, 2022-10-24:
@sacredx72 you should check if your device supports WebRTC: https://github.com/guino/Merkury1080P/issues/37#issuecomment-1256527596 (use one of the links and login using the qr scanner function in the phone app) -- if your camera is supported it should show a live view of the device on the computer -- it's no RTSP but it is a way to view the image on the computer (if your camera is supported).

sacredx72, 2022-10-26:
Yes, WebRTC is supported by the link you specified, but it's not pure WebRTS, but through an encrypted communication channel through the tuya server (this video stream cannot be integrated into a local video recorder

guino, 2022-10-26:
@sacredx72 there's not a lot we can do unless you hook up a SERIAL-TTL adapter or download the flash with a hardware programmer. You serial port is likely the four thru-holes on the top of your pictures (right above the QR code) -- that's where you'd connect a SERIAL-TTL adapter. Your flash chip is the black one labeled XMC above the USB plug -- that's where you'd connect your flash programmer but my experience is that you won't be able to read/write it without removing it from the board (or at least disconnecting pin 6). If you have specific questions let me know and I'll do what I can to help -- just can't do much myself without the device.

sacredx72, 2022-10-27:
[doorbellW2.zip](https://github.com/guino/BazzDoorbell/files/9876012/doorbellW2.zip)

here is my dump file from flash, surprisingly, without 6 CLK he did not want to shoot the whole
what information else should help in unlock?

guino, 2022-10-27:
@sacredx72 your dump file is good. This is a MIPS processor running what seems to be a linux application called 'ubia_t31' which I have never seen. I quickly looked into it with Ghidra and it doesn't seem to have any RTSP/ONVIF or even web services for that matter. None of the stuff I've seen on other devices with SD card tricks seem to be present on this application either. So really there's no easy way to get any 'direct' RTSP stream from this device. I also could not find any reference to telnet support either.

You could potentially modify the main application (i.e. inject a call to an application on the SD card) but it would definitely not be for enabling RTSP or any kind of streaming -- at most perhaps enable some logging and get notifications from it but that would probably require some coding (not scripting).

You may want to take a look at the information here: https://www.reddit.com/r/homeassistant/comments/s6i27l/i_managed_to_forward_the_tuyasmart_life_webrtc/ -- someone claimed to have used the WebRTC interface to generate a RTSP stream from it -- somoene with a little more time could probably make this into an easy to use tool for any tuya camera supporting WebRTC (most newer devices seem to support it).