guino/BazzDoorbell #89: Lost RTSP stream, still can access the telnet

sinclairfr, 2022-10-23:
Hello,

My hacked 8S doorbell was working fine for months but all of a sudden I lost RSTP feeds. 
The telnet access remains OK, the snap.cgi and mjpeg feeds are working fine.
I cannot access it from the tuya app anymore, it shows as offline.
RTSP : timeout.

I made sure that the SD card was not full and flushed the /mnt/mmc01/SDT folder.

It appears the doorbell now thinks were 1970.

Could a date problem cause all of the misfunctions? How is the date set in the first place? NTP server ?
Thanks

guino, 2022-10-23:
@sinclairfr the date is not an issue as many people use the device off-cloud, but it is an indication that it can’t connect to the tuya servers to sync the time — did you make any changes to router/firewall settings at home? If the device doesn’t sync the time with the tuya servers it won’t startup completely so it won’t show in the app and won’t enable rtsp unless you used an offline patch.

sinclairfr, 2022-10-23:
Thanks for your prompt answer! 
It is strange because I did not touch anything. I've tried rebooting several times, but Nothing helps.
Perhaps I can't access the tuya servers for some reason.
I've tried setting up the date in telnet but it does not seem to hold it after a restart.

Perhaps I should hard code the date somewhere in the files just to give it a test...

guino, 2022-10-24:
@sinclairfr Setting the date manually will not help in any way. Perhaps the device got removed from the tuya cloud and you may need to re-enroll it (you should not need to hack it again). If there’s an offline patch for your firmware that would allow RTSP to start without the need to connect to the tuya cloud (but the phone app won’t work). 

The date on these devices is never persistent - it always relies on the internet to set the date and time (and that’s why the tuya application doesn’t start completely as it waits for a connection to set the date and time before starting).

sinclairfr, 2022-10-24:
Thanks for the insights, much appreciated. 👍 
I found a perfect copy of the working SD card and went ahead and boot with it. It boots, I have snap.cgi, mjpeg and all but no RTSP and 1970 for the date.
This is definitely a problem stemming from elsewhere, the cloud probably...
As mentioned in the discussion [https://github.com/guino/BazzDoorbell/issues/77](url) , I used the #2 method to hack my doorbell camera.
How do I go about resetting the camera to the original firmware so I can link it to the tuya cloud again? (it does not boot with SD card because I used #2 method)
Thanks for your help!

sinclairfr, 2022-10-24:
For some reason the output of [ http://admin:056565099@ip/proc/self/root/etc/init.d/S90PPStrong ](URL) is now the content of the file, just as if it was not executable :

`#!/bin/sh

export PATH=/usr/bin:/sbin/:/usr/sbin:/bin

RED="[1;31m"
NORMAL="[0;39m"

echo "${GREEN} 2015 PPStrong Tech Cop.Ltd.${NORMAL}"

mkdir -p /opt/pps
MTDNUM=`cat /proc/cmdline | sed 's/.*ppsAppParts=\([0-9]\).*/\1/'`

# debug
#MTDNUM=5

echo "------------->mtdnum:${MTDNUM}"

case $MTDNUM in
         5)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
			break
			;;
		 7|8)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	break
		 	;;
		 0)     
			sleep 10
			mount -t vfat /dev/mmcblk0p1 /opt/pps
			break
			;;
		 *)
			MTDNUM=5
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	;;
esac

echo "/opt/pps/" > /tmp/PPStrong.runpath
[ -e /opt/pps/initrun.sh ] && cp /opt/pps/initrun.sh /tmp/PPStart && chmod +x /tmp/PPStart && /tmp/PPStart
`

closed 2022-10-24 by sinclairfr

sinclairfr, 2022-10-24:
I was able to re-pair the device to the tuya app without changing any file, I just held the reset button while restarting while having the QR code ready on the tuya app. After many tries, it did work. I just redid the hack and everything seems to work, I have the correct date. The RTSP feed does not work but I'm guessing that's because the password has changed.

guino, 2022-10-24:
@sinclairfr older devices don't usually have RTSP passwords -- some of the newer onvif devices having rtsp on rtsp://IP:8554//Streaming/Channels/101 need user admin and password admin (or may be set in tuya_config.json). It does seem like your issue was likely just something botched in the tuya servers where your device got removed from their cloud server and you had to re-enroll it.

sinclairfr, 2022-10-24:
Indeed, that's the same conclusion I reached, my device must have gotten kicked out of the database. 
The RTSP does work now. Indeed, no password is required.
A million thanks for your help.