guino/BazzDoorbell #91: "Bell J1" doorbell - Bell 7T/7S v3.1.2

peterstrapp, 2022-11-29:
I've just purchased a ["Bell J1" doorbell](https://www.amazon.co.uk/gp/product/B08GCG14G7/) with the hope of obtaining access to an RTSP stream, via a modified ppsapp. However it doesn't have a 2.x or 4.x firmware and I've been unable to fetch kernel args from '/proc/cmdline', so I'm hesitant to try guino's exploit. It identifies itself as a Bell 7T or Bell 7S. Has anyone else any experience with this device? 

Current version:  3.1.2.20220720
Update available: 3.1.2.20220922

By default port 80 is open but does not respond. After adding the below `ppsFactoryTool.txt` to an SD card the device starts up with a webserver running on port 8090. The deviceinfo endpoint is then accessible using the username `PpStRoNg` and password `#`. However the other endpoints ("/proc/*, etc)" return a 404.

ppsFactoryTool.txt
```
module=wifi,,ssid=Grivet,,password=123abc123def,,
```

http://PpStRoNg:#@192.168.1.121:8090/devices/deviceinfo

```
{
	"devname":	"Smart Home Bell",
	"model":	"Bell 7T",
	"serialno":	"xxxxxxxxx",
	"softwareversion":	"3.1.2",
	"hardwareversion":	"BE7S_T1_V10",
	"firmwareversion":	"ppstrong-b81-m_neutral_std-3.1.2.20220720",
	"licence_id":	"xxxxxxxxxxxxxxxxxxxxx",
	"licence_key":	"xxxxxxxxxxxxxxxxxxxxxxxxxx",
	"identity":	"XXXXXXXXXXXXXXXX",
	"WiFi MAC":	"xx:xx:xx:xx:xx:xx",
	"ETH MAC":	"xx:xx:xx:xx:xx:xx",
	"mcuversion":	"meari-2.4.1.20220712-bell7s"
}
```
There's an unpopulated 6 pin header and 4 pads that _may_ be a UART. It appears to be based on a Ingenic T31 ZL and has a Hi3861 controller, possibly for wifi.

[image]

[image]

peterstrapp, 2022-12-01:
It looks like the 3.1.2 firmware probably is Linux based...

https://meari-eu.oss-accelerate.aliyuncs.com/firmware/ppstrong-b81-m_neutral_std-3.1.3.20220922-upgrade.bin

```
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
1442208       0x1601A0        uImage header, header size: 64 bytes, header CRC: 0xE41A3FB9, created: 2022-09-19 03:58:49, image size: 2557952 bytes, Data Address: 0x80010000, Entry Point: 0x803DA400, data CRC: 0xBD6CAF7, OS: Linux, CPU: MIPS, image type: OS Kernel Image, image name: "Linux-3.10.14-Archon"
1442272       0x1601E0        LZO compressed data
3471187       0x34F753        lzop compressed data,d52,
4001312       0x3D0E20        LZMA compressed data, properties: 0x6E, dictionary size: 65536 bytes, uncompressed size: 700176 bytes
4432240       0x43A170        Squashfs filesystem, little endian, version 4.0, compression:xz, size: 1887938 bytes, 45 inodes, blocksize: 131072 bytes, created: 2022-09-22 02:11:04
7017827       0x6B1563        SHA256 hash constants, little endian
7018150       0x6B16A6        AES S-Box
7018464       0x6B17E0        AES Inverse S-Box
7287443       0x6F3293        Unix path: /home/fangbiao/platform(#
7305524       0x6F7934        Certificate in DER format (x509 v3), header length: 4, sequence length: 831
7313618       0x6F98D2        AES S-Box
7320533       0x6FB3D5        AES Inverse S-Box
7321981       0x6FB97D        Base64 standard index table
7344399       0x70110F        SHA256 hash constants, little endian
7851176       0x77CCA8        AES S-Box
7851491       0x77CDE3        AES Inverse S-Box
7851747       0x77CEE3        Base64 standard index table
```

Contents of firmware upgrade squashfs image:
```
.
├── bin
│   ├── devcmd
│   ├── ppsconfig
│   └── pps_dlink
├── init
│   ├── initrun.sh
│   └── udhcpc.script
├── ko
│   └── pps_devlink.ko
├── lib
│   ├── libagc.so
│   ├── libalog.so
│   ├── libaudioProcess.so
│   ├── libgcc_s.so
│   ├── libgcc_s.so.1
│   ├── libhpf.so
│   ├── libimp.so
│   ├── libjzdl.m.so
│   ├── libns.so
│   ├── libppsmedia_barcode_hd.so
│   ├── libppsmedia.so
│   ├── libppsnn.so
│   ├── libpps_plat_pd.so
│   ├── libpps_storage.so
│   ├── libsrtp2.so
│   ├── libstdc++.so
│   ├── libsysutils.so
│   ├── libWebrtcClient.so
│   └── libWebrtcUtils.so
├── logo
│   ├── arenti-2304x1296.bmp
│   ├── arenti-640x360.bmp
│   ├── vacos_logo.bmp
│   └── vacos_logo_sub.bmp
├── models
│   ├── OCclPT80WRs3CQMrJiE-CyM6LRI5NSgSLiNUYjgkcwsgAloNYDNcRmYbfkp8eiseIw.bin
│   └── OCclPT80WRs3CQMrJiE-CyM6LRI5NSgSLiNUYjgkcwsgAloNYDNdQWAbeUl6C3tHf1t0RnxhE188KA.bin
└── sound
    ├── dingdong.wav
    ├── login.wav
    ├── restart.wav
    ├── warning.wav
    ├── wifi_psk_err_cn.pcm
    └── wifi_psk_err_en.pcm
```

guino, 2022-12-01:
@peterstrapp The few 3.x firmware I have seen were not running full linux (was some RTOS based stuff). In the upgrade file you provided it does seem like the kernel and main application is bundled into a single block of memory (RTOS style) as binwalk didn't even decompress it.

I would be willing to bet this is a battery-operated device and running linux on battery power would just not be efficient, so they use something they can put on deep-sleep and wake up on demand with sensors and such, so streaming from it for a long time would definitely drain the battery fast.