guino/BazzDoorbell #95: LSC Smart Doorbell login and password to ONVIF

KapitanKapitan, 2023-02-19:
Hello,

I try to connect this device to my NVR by ONVFI.

Settings in app:
[image]

Settings in NVR:
[image]

Login: admin
Password: admin

But I get password error on my NVR or i7-VMS Lite software. 

What I do wrong? What is the correct login data?

KapitanKapitan, 2023-02-19:
http://192.168.18.156:8090/search

display this

{
	"sn":	"102204378",
	"factory_code":	0,
	"factory_code_str":	"",
	"model":	"Bell 8S",
	"tp":	"lasivvb8ccnsma4t",
	"p2p_uuid":	"",
	"ip":	"192.168.18.156",
	"mask":	"255.255.255.0",
	"gw":	"192.168.18.1",
	"mac":	"b4:fb:e3:d9:ef:65",
	"interface":	"wlan0",
	"version":	"4.0.7"
}

guino, 2023-02-20:
@KapitanKapitan many devices have the onvif password defined in tuya_config.json -- check for settings under http://admin:056565099@ip/proc/self/root/home/cfg/tuya_config.json as that may have the onvif password if it's different than the default one.

KapitanKapitan, 2023-02-20:
Thx for you response, but I have another problem, when I put http://admin:admin@192.168.18.156:8090/proc/self/root/etc/init.d/S90PPStrong

i still get 
`# debug
MTDNUM=5`

my http://192.168.18.156:8090/proc/cmdline is:
`setenv bootargs mem=64M console=ttySAK0,115200n8 loglevel=10   mtdparts=spi0.0:256k(bld),64k(env),64k(enc),64k(sysflg),3m(sys),4032k(app),640k(cfg) ppsAppParts=5 ip=192.168.1.99:::255.255.255.0 eth=00:55:7b:b5:7d:f7 ethDisable`

So I prepare env file in nothepad like this:
`bootargs=mem=64M console=ttySAK0,115200n8 mtdparts=spi0.0:256k(bld),64k(env),64k(enc),64k(sysflg),3m(sys),4032k(app),640k(cfg) ppsAppParts=5 ppsWatchInitEnd - ip=\\${T//_/\\$\\'"\\\\x20"\\'}:::::";T=\\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\\";eval"
 `

But still I get:
``# debug
MTDNUM=5``
without # before MTD....

Can you help me with that?

KapitanKapitan, 2023-02-20:
So form Iv get:

`{
	"version":	1,
	"sleep_mode":	0,
	"alarm_fun_onoff":	0,
	"alarm_fun_sensitivity":	1,
	"alarm_fun_mode_switch":	0,
	"alarm_fun_time_start":	0,
	"alarm_fun_time_end":	0,
	"flip_onoff":	0,
	"light_onoff":	1,
	"night_mode":	0,
	"sound_detect_onoff":	0,
	"sound_detect_sensitivity":	0,
	"watermark_onoff":	1,
	"event_record_time":	60,
	"enable_event_record":	2,
	"record_enable":	1,
	"motion_trace":	1,
	"motion_area_switch":	0,
	"motion_area":	"",
	"motion_tracking":	0,
	"cry_detection_switch":	0,
	"humanoid_filter":	0,
	"loudspeaker_vol_pct":	100,
	"jingle_mode":	2,
	"jingle_sound":	1,
	"jingle_volume":	100,
	"jingle_exist":	0,
	**"onvif_enable":	1,
	"onvif_pwd":	"123456"**
}
`

So looks like ONVIF is enabled and password is 123456. But still I get error wrong password...

guino, 2023-02-21:
@KapitanKapitan the `#` in front of MTDNUM will **NOT** be removed. The important thing is that after you follow the steps the response to http://192.168.18.156:8090/proc/cmdline changes (it should end with a section like `ip=\\${T//_/\\$\\'"\\\\x20"\\'}:::::";T=\\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\\";eval"`).

If you have onvif password 123456, the user should be admin and it should just work. Have you tried it with different software (i.e. phone, computer, etc) -- I would verify it works 'anywhere' before trying to set it up with the NVR.