guino/BazzDoorbell #99: adjust to DST

bkbartk, 2023-03-28:
Hello,

does anyone know how to adjust the timestamp in the video to DST?
6 months back, I remember after a lot of trying I was able to adjust this,
but I forgot, and didn't document.

bkbartk, 2023-03-28:
ok, found something,
If I disable offline.sh and also all other blocking.
the file
`/home/cfg/TZ`
changes from
`CST-1:00:00`
to
`CST-2:00:00`

and that's it,
only when changing this manual I need to apply a reboot.
would there be a way to do this without any reboot?

guino, 2023-03-28:
@bkbartk when using the camera online the tuya servers adjust the time automatically. When using the device offline you may need to adjust the time on your ntp server or device (as you pointed out) -- since that timezone file is likely only read when starting the main application you will need to either reboot or kill ppsapp and run it again.

closed 2023-03-28 by bkbartk