guino/BazzDoorbell #101: only 1090x1080 with RTSP

stack0423, 2023-04-12:
I have only 1090x1080 RTSP
How can i version like 720?

guino, 2023-04-12:
@stack0423  Some devices only have 1 RTSP stream (i.e. rtsp://IP:8554 ) and some devices have 2 RTSP streams (i.e. rtsp://IP:8554//Streaming/Channels/101 and rtsp://IP:8554//Streaming/Channels/102 ). The 'main' RTSP stream is always the same as the full resolution of the device. The secondary stream (102 -- if available) is a lower resolution stream which I believe is around 360p), so that would also not give you 720p but may be enough for you.

If you 'require' exactly 720p the only solution would be to convert the RTSP feed from 1080p to 720p with something like ffmpeg (which you would have to run on a separate device like a computer, pi, etc).

stack0423, 2023-04-12:
I put the stream to motioneye but the stream is not smooth. can i do something with settingg in the sofware?

guino, 2023-04-12:
@stack0423 I see that with all tuya devices: when motion detection is triggered it takes 'extra processing' from the board and it can cause some frame skipping (even the recorded video can miss frames). You can try to disable recording and/or motion detection in the device (and record the RTSP stream on your RTSP client side) to see if it helps. You could also try to record 24/7 instead of event recording, but that will probably wear out your SD card and not be as fast as disabling recording altogether.

If you're getting frame skipping when there's no motion detection going on: that is usually an issue with wifi/signal (i.e. congested 2.4Ghz band) in which case you could try moving your router/antenna and/or switching wifi channel on your router to see if it helps.

stack0423, 2023-04-13:
I have the bell blockt to outside ,no connection to tuya server. 
And i have logpaser mqtt to my domoticz, for motion detection and somewone push the doorbell.