guino/BazzDoorbell #102: Question - Motion detection event

adjivas, 2023-04-15:
Hello,

Thank you for this great project

I need your advice on something.
I'd like to implement an MQTT event produced by the detection of a motion.

I have a Mini 16S patched with an offline ppsapp file (see [323fc1b9791a6dcd37ceabb849f120f4](https://github.com/guino/ppsapp-rtsp/issues/1)
The capture MJPEG video works

To have this feature, my plan is to write a demonized Mini 16S's shell script
It will have this instructions: 
* Get the current image with a command like `curl -u "user:password" -o /tmp/test.jpg localhost/cgi-bin/snap.cgi` 
* Get the value (an float) difference between two last images with a command like `magick compare -metric rmse test.jpg test.jpg NULL:`, 
* Depending of the result, it will send an adapted event with a mqtt_pub command like `mosquitto_pub -h "192.168.0.X" -m "detected" -t home/mini16s/motion`

What I will need is to have a ARMv5 version of the curl and magick binaries.
The ARMv5's mqtt_pub has been already shared by this helpful comment [1046337300](https://github.com/guino/BazzDoorbell/issues/35#issuecomment-1046337300)

Do you think this is the correct solution?

guino, 2023-04-15:
@adjivas you really don’t want to do that in the device itself (it is just not powerful enough to do it). You could do that on a separate machine (get the image, compare and generate an event when required), but what most people do is use the built-in motion detection from the main application to generate the mqtt messages from the device itself. This should work offline just the same buy you may need to put it online temporarily to enable motion detection settings in the phone app (or you can try to enable it in the tuya_config.json file directly). 

To get the motion detection events from the main application you just use the posted log_parser script. 

Please see the below for further information:
https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Integrate-with-Home-Assistant,-HomeBridge,-Domoticz,-etc
https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Use-notifications-MQTT-log_parser-without-a-patched-ppsapp-%3F

adjivas, 2023-04-15:
@guino Thanks a lot for your answer and your links :)

closed 2023-04-15 by adjivas