guino/BazzDoorbell #105: no stream, no telnet need a little help

FritsOV, 2023-05-12:
Hello, 
I'm struggeling with my LSC doorbel. Over a year ago i (think) i patched the doorbel. But i can't 
1. get into it with telnet with an error connection refused
2. after http://192.168.1.69:8080/cgi-bin/snap.cgi i am asked to put in a password. admin, admin  and admin 056565099 gives me connection refused
3. i can see that the onvif is disabled with password admin 
4. i cant see anything picked up in the mqqt log files for movement or puss button.
5. On a pi where tuya is installed i can see the doorbell and have a stream. 

i'm totally out of clues and dont know where to start, its a bit of messy (sorry) in BazzDoorbell github. 

Any help would be apreciated

Kind regards,
Frits

Device info:
<html><body>

devname | "Smart Home Camera"
-- | --
model | "Bell 8S"
serialno | "061063853"
softwareversion | "2.10.5"
hardwareversion | "BE8S_H1_V10_433"
firmwareversion | "ppstrong-c51-tuya2_lcs-2.10.5.20210806"
authkey | "Yt9U6ApicBkiQ1ADuJrX5pbk9FyzB056"
deviceid | "pp016274578bf16ddec6"
identity | "MR2007180100101886"
pid | "aaa"
WiFi MAC | "d4:d2:d6:d9:c2:3c"

</body>
</html>

**cmdline:**
mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=0 ppsWatchInitEnd ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuno,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,T mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts

**[Tuya](http://192.168.XX.XX/proc/self/root/home/cfg/tuya_config.json)**
{
	"version":	1,
	"sleep_mode":	0,
	"alarm_fun_onoff":	1,
	"alarm_fun_sensitivity":	0,
	"alarm_fun_mode_switch":	0,
	"alarm_fun_time_start":	0,
	"alarm_fun_time_end":	0,
	"flip_onoff":	0,
	"light_onoff":	1,
	"night_mode":	0,
	"sound_detect_onoff":	0,
	"sound_detect_sensitivity":	0,
	"watermark_onoff":	1,
	"event_record_time":	60,
	"enable_event_record":	1,
	"record_enable":	0,
	"motion_trace":	1,
	"motion_area_switch":	0,
	"motion_area":	"",
	"motion_tracking":	0,
	"cry_detection_switch":	0,
	"humanoid_filter":	1,
	"loudspeaker_vol_pct":	90,
	"jingle_mode":	0,
	"jingle_sound":	1,
	"jingle_volume":	100,
	"jingle_exist":	0,
	"flight_bright_mode":	0,
	"flight_light_brightness":	100,
	"flight_pir_set":	0,
	"flight_pir_one":	0,
	"flight_pir_two":	0,
	"flight_pir_three":	0,
	"flight_pir_sensitivity":	0,
	"flight_alarm_fun_onoff":	0,
	"flight_on_off":	0,
	"flight_pir_light_on_time":	30,
	"flight_warn_switch":	0,
	"flight_dualbrite":	0,
	"flight_ontime":	0,
	"flight_highbrightess":	10,
	"flight_lowbrightess":	10,
	"flight_mode":	0,
	"flight_motion_sens":	0,
	"onvif_enable":	0,
	"onvif_pwd":	"admin"
}

**http://192.168.1.69/proc/self/root/etc/init.d/S90PPStrong**

#!/bin/sh

export PATH=/usr/bin:/sbin/:/usr/sbin:/bin

RED="[1;31m"
NORMAL="[0;39m"

echo "${GREEN} 2015 PPStrong Tech Cop.Ltd.${NORMAL}"

mkdir -p /opt/pps
MTDNUM=`cat /proc/cmdline | sed 's/.*ppsAppParts=\([0-9]\).*/\1/'`

# debug
#MTDNUM=5

echo "------------->mtdnum:${MTDNUM}"

case $MTDNUM in
         5)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
			break
			;;
		 7|8)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	break
		 	;;
		 0)     
			sleep 10
			mount -t vfat /dev/mmcblk0p1 /opt/pps
			break
			;;
		 *)
			MTDNUM=5
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	;;
esac

echo "/opt/pps/" > /tmp/PPStrong.runpath
[ -e /opt/pps/initrun.sh ] && cp /opt/pps/initrun.sh /tmp/PPStart && chmod +x /tmp/PPStart && /tmp/PPStart

guino, 2023-05-13:
@FritsOV 
For telnet you should at least get a login prompt -- if not: check the IP address and check your busybox file in the SD card (it should be over 1MB in size).

For snap.cgi (http://192.168.1.69:8080/cgi-bin/snap.cgi) the password is whatever is in your httpd.conf file (SD card) -- the default is user:password -- please double check you have configured the address in snap.cgi to be 0x056e6b0 (for your firmware).

Onvif will require you to edit tuya_config.json to set onvif_enable to 1 (yours is disabled right now) -- some details here: https://github.com/guino/BazzDoorbell/issues/2#issuecomment-745618597 . You could try copying the ppsapp from the home directory to the root of the SD card and removing the '#' in front of the line `#/mnt/mmc01/set onvif_enable 1` in custom.sh but some devices have issues with killing ppsapp and will go into boot loop (in which case you have to remove ppsapp from the root of SD card and manually edit tuya_config.json).

Since your device doesn't have a patched ppsapp, you'll need to follow this to get MQTT notifications working: https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Use-notifications-MQTT-log_parser-without-a-patched-ppsapp-%3F

If you installed tuya on a Pi I assume you're using an android image like lineage OS ? I've never tried it but it should be the same as using it on your phone -- this only helps to confirm the device is online and working.

FritsOV, 2023-05-15:
Thank you for your responce that is highly apreciated!

I will follow your leads and give in advance some extra info. i hope to get it working this week. 

- I am running HAS on a raspberi pi 4 with 8 gig.  i should have explained this, and telling that i can see the doorbell in the HAS Tuya add on. The HAS image is the one from the install page from homeassistant. 
- I was in the consumption that i had patched the doorbell,. In the patch sequence i got at some point the "done" page. 

i will give some feedback after i followed your pointers. 

Thanks!
FritsOv

FritsOV, 2023-05-17:
Hello Guino,

Been working over a day on the doorbell and here is what i got so far lets start with the good news is that some parts are working the bad news is i do not have the propper folderstructure on the SD-card. Thats i guess why telnet and all other features are not working correct. 
Could you help me out with:
- getting the propper structure for SD card (see below)
- MQTT

**What is working:**
- snap.cgi and mjpeg.cgi is working. 
- https://www.youtube.com/results?search_query=go2rtc+lsc+camera+doorbell -> gives me done as result in the browser 
- doorbell without an SD card wont boot, it only boots with SD card on it so in combination with all the above i would say it is patched but i dont have for some reason another problem (it was 2 years ago that i patched the device)

**What is not working:**
- still can't get into the device with putty on port 22. when i start putty and it tries to connect to the device it does not even come to the point (after about 30 seconds) to ask user and password but gives as result: connection refused. I think this is because off the next point:
- On my SD card there is only two folders i.e. cgi-bin and STD. all the files are in the root directory. I guess there went something wrong and i was wondering if there is a way to get the propper folderstructure as shown in the picture [here](https://github.com/guino/BazzDoorbell/issues/39#issuecomment-887776786) folders for the firmware 2.10.5 BE8S_H1_V10_433. 

my file structure is:
 Map van F:\

27-02-2022  10:27    <DIR>          cgi-bin
16-05-2023  14:28             1.085 custom.sh
27-02-2022  20:26               285 hosts
15-05-2023  17:24                16 httpd.conf
27-02-2022  20:26             1.327 index.html
27-02-2022  20:26             7.956 jpeg-arm
27-02-2022  20:26               131 mqtt_pub
27-02-2022  20:26             1.064 offline.sh
27-02-2022  20:26                37 passwd
27-02-2022  20:26               263 set
27-02-2022  20:26               161 upload.html
12-05-2023  13:08         1.109.128 busybox
11-05-2023  17:19                57 ppsFactoryTool.txt
05-12-2020  06:47               102 ppsMmcTool.txt
16-05-2023  14:00               926 env
27-12-2020  13:13             1.108 initrun.sh
16-05-2023  00:06    <DIR>          SDT
27-02-2022  20:26             1.785 doorcopy.sh
27-02-2022  20:26               131 dropbearmulti
27-02-2022  20:26            24.729 README.md
08-08-2022  14:25            40.056 reredirect
07-12-2020  18:45            75.220 mosquitto_pub
16-05-2023  15:11             1.662 old_log_parser.sh.txt
16-05-2023  21:19             1.030 log_parser.sh
17-05-2023  10:48                 0 filestruct.txt
              23 bestand(en)        1.268.259 bytes
               2 map(pen)  31.850.430.464 bytes beschikbaar

Despite the fact that it has no use to proceed further i started to look at MQTT:
i made a new user doorbell, doorbell (change that later if it works). And 
changed the logparser.sh lines:

	if contains "$BUF" "--motion detection alarm --"; then
	    #echo "motion detected"
            /mnt/mmc01/mosquitto_pub -h 192.168.1.65 -u doorbell -P doorbell -m "detected" -t home/doorbell/motion
	elif contains "$BUF" "##doorbell_push 3"; then
	    #echo "doorbell push button"
            /mnt/mmc01/mosquitto_pub -h 192.168.1.65 -u doorbell -P doorbell -m "pushed" -t home/doorbell/button
	#else
	    #echo "Unknown cmd: $BUF"

In  MQTT i can see that the doorbell is ringing (detection wont work) but the line in MQTT log disapears 
positive is that MQTT is picking up the responce  but  there is no home/doorbell/ in the folderstructure. So that will never work. 

MQTT Log file:
2023-05-16 21:24:55: New connection from 192.168.1.69:56306 on port 1883.
2023-05-16 21:24:55: New client connected from 192.168.1.69:56306 as mosq-2ada4vgCpcW2SwCqAC (p2, c1, k60, u'doorbell').
2023-05-16 21:24:55: Client mosq-2ada4vgCpcW2SwCqAC disconnected.
2023-05-17 10:34:42: Client auto-767B90EF-98B3-8DBA-B019-B8A9B73C8251 disconnected.
2023-05-17 10:34:42: New connection from 192.168.1.69:51792 on port 1883.
2023-05-17 10:34:42: New client connected from 192.168.1.69:51792 as mosq-rPodfScRJSgkuvvFde (p2, c1, k60, u'doorbell').
2023-05-17 10:34:42: Client mosq-rPodfScRJSgkuvvFde disconnected.
2023-05-17 10:34:43: New connection from 192.168.1.69:51796 on port 1883.
2023-05-17 10:34:43: New client connected from 192.168.1.69:51796 as mosq-vo86RbyadR1hamf34h (p2, c1, k60, u'doorbell').

FritsOV, 2023-05-17:
MQTT part is working now.
I made an automation witch activates when the doorbell is triggerd so that part works also. 

If you could give me some help with the telnet and the filestructure that would be verry helpfull.

Kind regards and thanks in advance for your time and effort. 

FritsOv

guino, 2023-05-17:
@FritsOV do you get a prompt when you do 'telnet IP' from the computer ? if you get a prompt but can't login, you just need to configure the passwd file as defined in step 9 of : https://github.com/guino/BazzDoorbell/issues/2 

If you don't get a prompt then I would delete the busybox file from the SD card and download it again from here: https://github.com/guino/BazzDoorbell/blob/master/mmc/busybox?raw=true -- please make sure you 'eject' the SD properly before removing it from the computer otherwise the file may get corrupted and it won't work.

It is also possible that your hardware has an older hardware and may require an older busybox compatible with it, you can try this one if the one above still doesn't work: https://github.com/guino/Merkury720/raw/main/mmc/busybox

guino, 2023-05-17:
@FritsOV I just noticed you mentioned using putty on port 22 -- that's SSH (NOT telnet). You can use putty with telnet mode but it should be port 23 (telnet).

Once you get telnet working you can copy the home, bin and lib directories copied. In any case, the initrun.sh should extract these folders for you IF there's no 'home' folder in the SD card already.. this is the line doing it:
`if [ ! -e /mnt/mmc01/home ]; then tar xzf /opt/pps/app.tar.gz -C /mnt/mmc01/; fi`

I have seem some newer firmware with a slightly different packaging in which case we had to change it to be this:
`if [ ! -e /mnt/mmc01/home ]; then tar xf /opt/pps/app.* -C /mnt/mmc01/; fi`

You should be able to try the above without any risks.

FritsOV, 2023-05-17:
Thank you  for your time and tips!
I have tried on putty with port 23 in raw mode an get to the point asking for user and login. user:password is not accepted, (thats in the passwrd file). 
telnet gives me the same result 
for argument sake i tried admin admin  and admin 056565099 unfortunatly all with the same result. 

Tomorrow i wil try your sugestion with the bussybox and i even could try to open up the device to look inside and provide a picture. That will take me some more time. 

I'll be back

guino, 2023-05-18:
If you get a login prompt then your busybox is fine, you just need to configure telnet password (or passwordless telnet) as described in step 9 of #2

FritsOV, 2023-05-19:
Finally!  after telnet option -l /bin/sh added i can telnet and modified the initrun.sh uncommenting the line: 
if [ ! -e /mnt/mmc01/home ]; then tar xzf /opt/pps/app.tar.gz -C /mnt/mmc01/; fi
the propper foderstructure is on the SD card now.
i understand what is ment with getting the md5 but I dont know what exactly is ment with adding hash  and where to put is.

patched the rtcp ppspap and this also works now perfectly, i can see the stream in vlc and in the onvif aplication. 
json output tells me that onvif is enabled. and all works on port 8554.

will try the ssh option in your wiki and if all works i will consider the option of the newer patch as sugested above. 
The question that remains is can i cut the device off for connecting to the cloud without the newer patch like as it is now or is 
patching to [#13](https://github.com/guino/BazzDoorbell/issues/13) mandatory? 
Many thanks  will buy you a beer... ;)

guino, 2023-05-19:
@FritsOV you don’t need to use #13 to cut off cloud connection (offline) - you can do it with your existing setup. 

This is the information you need to use it offline: https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Use-the-device-OFFLINE-(disconnected-from-tuya-servers)-%3F

if you need/want to use an offline patch I will need a copy of your ppsapp file so I can make a patch for it.

Thanks for the 🍺!

FritsOV, 2023-05-22:
> @FritsOV you don’t need to use #13 to cut off cloud connection (offline) - you can do it with your existing setup.
>
> This is the information you need to use it offline: [...]

Hello Guinno,

Sorry for the late responce, had som obligations. 
I would love to send you the ppsapp file. but first i need to get 2 things working.
1. telnet and md5 :  do i need to set in the passwd file  /<user>:<passw>  <md5 number> ?
2. have to take care of the ssh part. hope to do this in some spare time. 
3. then i can proceed with the next step in #13. 

give me a few days.

guino, 2023-05-23:
@FritsOV I've been traveling myself, so haven't had time to check on github much.
1-For telnet you just need to configure the passwd file as defined in step 9 of : https://github.com/guino/BazzDoorbell/issues/2
2-SSH is optional -- I don't use it, don't recommend it as you can do anything you want without it (including upload/download files)
3-Since you have #2 working, I would only use #13 if you want/need to boot the device without the SD card (everything else is the same).