guino/BazzDoorbell #106: LSC Bell 8T

yoyomax80400, 2023-05-13:
Hi guy,

I already have the onvif port open trough the tuya app.
I want to get the info when the button is pushed, cause actually I don't have any idea how to ... 

Port 8090 is open but information is really different : 
**/devices/deviceinfo :** 
{"devname":"Smart Home Camera","model":"Bell 8T","serialno":"100428979","softwareversion":"5.0.6","hardwareversion":"BE8T_S1_SC3_433","firmwareversion":"ppstrong-c71-tuya2_lsc-5.0.6.20210331","identity":"MR2011031400907617","authkey":"VNKAkgy3itwdFDfdoSdL34ArpbOxDNDn","deviceid":"pp01177fd2c15bee7feb","pid":"aaa","WiFi MAC":"b4:fb:e3:6a:9e:14","ETH MAC":"b4:fb:e3:6a:9e:14"}

**/proc/cmdline**
console=/dev/null LX_MEM=0x3fe0000 mma_heap=mma_heap_name0,miu=0,sz=0x1d00000 pcbversion=BELL5S_S1_V10 sensor=sc3335mipi

Could you please help me if it possible ?

guino, 2023-05-13:
@yoyomax80400 sorry but this device's cmdline indicates it is likely a RTOS image (not linux) -- I am taking a guess your device has a built-in battery in it ? 

The tools we have are only for linux devices and there will be very little you can customize on that device unless you want to directly alter the firmware and tweak the code here and there (with ghidra + hardware programmer). 

If it's a newer doorbell integrated to the latest tuya api you may be able to get notifications using the tuya plugin for scrypted (which you have to install+configure). I know my doorbell is too old for it but my newer cameras work with it I believe it had a way to provide motion notifications (so doorbell push notifications may be possible too).

yoyomax80400, 2023-05-13:
@guino Thanks for answer 
No the doorbell is made with too wire to supply it, no battery inside 
I can use USB too. 

I bought it the previous year in 2022, I already try to get information with DP ID from Tuya but unfortunately not working … 
Motion are ok from Tuya API but doorbell push not …

yoyomax80400, 2023-05-14:
@guino 
Any idea if we can contact the dev on the Tuya integration in scrypted ? 

Caisse it's seams the information exist when I push the button but is recognized as unknown : 

Received new TuyaPulsar Message:
- Unknown bizCode: event_notify with data: {"devId":"bf19d24b88919e7a73uyp2","edata":"617bdf7a1684051778","etype":"ac_doorbell"}.

Or do you have any idea how I could parse it ?

closed 2023-05-15 by yoyomax80400