guino/BazzDoorbell #109: Bell 15T any hope to eneble onvif/rstp ?

ippocratis, 2023-06-03:
I got a video door bell

Oem is cooau
https://www.cooau.com/products/bell-j6-plus

Probably same as
https://shop.tuyaoem.com/doorbellcam/tuya-2k-qhd-wire-free-battery-smart-video-doorbell

nmap to doorbell IP says all ports are closed

ppsFactoryTool.txt works to open 8090 port

`8090/tcp open  opsmessaging`

Some api endpoints work with 

User: PpStRoNg
Password: #%&wL1@*tU123zv

http://192.168.1.20:8090/devices/deviceinfo
Return

`{
	"devname":	"Smart Home Bell",
	"model":	"Bell 15T",
	"serialno":	"xxx",
	"softwareversion":	"3.1.3",
	"hardwareversion":	"BE7S_T1_V10",
	"firmwareversion":	"ppstrong-b81-m_neutral_std-3.1.3.20220922",
	"licence_id":	"ppslf45b17112284497e",
	"licence_key":	"xxx",
	"identity":	"NBV00676CS82100575",
	"WiFi MAC":	"xxx",
	"ETH MAC":	"xxx",
	"mcuversion":	"meari-2.4.1.20220921-bell7s"
}`

And http://192.168.1.20:8090/devices/settings

`{"device":{"name":"Smart Home Bell","longname":"Smart Home Bell","model":"Bell 15T","serialno":"xxx","tp":"150000225","pcbversion":"BE7S_T1_V10","software_version":"3.1.3","firmware_version":"ppstrong-b81-m_neutral_std-3.1.3.20220922","mcu_version":"meari-2.4.1.20220921"},"time_now":"2015-12-31T16:40:05Z","timezone":"EET02:00:00EEST03:00:00,M3.5.0,M10.5.0","video_mirror":0,"led":{"all":{"enable":1}},"wlan":{"ssid":"vippo","gateway":"192.168.1.254","signal":87,"ip":"192.168.1.20","mac":"xxx","mask":"255.255.255.0"},"sdcard":{"company":"Unkown","status":1,"fstype":1,"total_space":"1.860G","free_space":"1.857G","is_formted":0},"sd_event_store":{"duration":30,"enable":1,"continuity":1},"alarm_plan":[],"alarm_feq":0,"cloud_storage":{"enable":1},"day_night_mode":0,"antiflicker":2,"people_detect":{"enable":1,"bnddraw":0,"enable_day_filter":0,"enable_night_filter":0},"bell":{"pir":{"enable":1,"level":4},"volume":100,"relay_enable":1,"power":"battery","battery":{"status":"discharging","percent":95,"remain":2718},"workmode":2,"charm":{"enable":1,"volume":75,"repetition":1,"song":["song1","song2","song3","song4"],"selected":"song1"}},"roi":{"enable":0,"width":0,"height":0,"bitmap":""},"speaker":1,"microphone":1,"siren_timeout":30,"rec_audio_en":1,"cur_network_mode":1,"network_supported":0}`

Any ideas on what I could possible try to enable onvif/rstp on this?

Ideally I would like to have an option to grab the video feed, store events on a remote machine and get motion events notifications

Thanks
keep up on this amazing repo

guino, 2023-06-04:
@ippocratis unfortunately I have never seen a device with firmware 3.x running linux -- they all run RTOS so they can stay 'asleep' and be triggered when something happens (presence detection, button push, etc), usually because it's a battery operated device so it doesn't drain the battery as fast. I realize a lot of these devices have the option to be hard wired and I assume that's how you'd want to use ONVIF/RTSP (otherwise the battery would drain quickly). 

Being a tuya device you could check if it has WebRTC support: https://www.reddit.com/r/smartlife/comments/oyqvdv/webrtc_stream_terminal_for_tuya_smartlife_cameras/ 

If it has WebRTC support you should be able to constantly stream it from a 3rd device (Computer, Raspberry Pi, etc) and both record (continuously, motion, etc) and provide RTSP service using something like https://www.scrypted.app/ -- again all of this will be useless if you intend to use the device on batteries.

ippocratis, 2023-06-04:
It has a hardwire mount option that constantly powers the device yes

I was unable to connect to the tuya app

cloudwdge app works though
the cooau app looks like a clone of the cloudedge app tbh

I was unable to get the /proc/cmdline output from the browser and curl

guino, 2023-06-07:
@ippocratis /proc/cmdline is a linux thing, so not being able to get that is another sing that it is probably not running linux.