guino/BazzDoorbell #110: Need some help

xmow49, 2023-07-16:
Hi,
First thanks for your work in this repo. It's very helpful.
I have an IP camera that works on the Cloudege app too.
I followed your readme, and I successfully read the flash of the camera:
[firm.bin](https://github.com/guino/BazzDoorbell/files/12064596/firm.zip)
I tested binwalk: 
```
dorian@dorian-VM:~/bin$ binwalk -e -M ./firm.bin

Scan Time:     2023-07-16 18:43:02
Target File:   /home/dorian/bin/firm.bin
MD5 Checksum:  c6d8ecbca7ca8d64a59e6f8b2d5491cf
Signatures:    411

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
241364        0x3AED4         CRC32 polynomial table, little endian
7733248       0x760000        JFFS2 filesystem, little endian

```
But he found only one jffs2 partition that contains cameras sound (parring) : 
```
dorian@dorian-VM:~/bin/_firm.bin.extracted$ cd jffs2-root/
dorian@dorian-VM:~/bin/_firm.bin.extracted/jffs2-root$ ls
ASC16  sound
dorian@dorian-VM:~/bin/_firm.bin.extracted/jffs2-root$ ls sound/
login.wav  restart.wav
dorian@dorian-VM:~/bin/_firm.bin.extracted/jffs2-root$
```
I used https://binvis.io/ to see the bin:
[image]
So, there is a partition at 0x70000. With an hex dump, i can see: `gziphead`
[image]

So I tested to create a .bin file with only this partition, and dezip:
```
dorian@dorian-VM:~/bin$ gzip -d test.gz

gzip: test.gz: not in gzip format

dorian@dorian-VM:~/bin$ file test.gz
test.gz: data
```
But no result.
Now I am stuck with this.
Do you have any idea of what is this partition, and how to decode it?

Thanks for your time.

guino, 2023-07-16:
@xmow49 jffs2 is a file system, and the extracted contents are likely correct. Since it didn't extract anything else, chances are you're dealing with a non-linux firmware (i.e. RTOS) where the OS+main application are combined into a single 'chunk' of memory that is loaded/executed. If you know the load address you may be able to use ghidra to decompile the code and see what functions are available in the code and perhaps enable some of it but you'll need to flash the changes back for any attempts and your options will be very limited. If the device was running linux, binwalk should have extracted boot scripts and the main application for the device. It is also possible for the main chunk to be compressed with some custom tool which would make it very difficult to make changes to it.

It is highly unlikely that anything you place/change in the jffs2 will make any difference in behavior of the device (perhaps the two sounds, but that's it).

xmow49, 2023-07-17:
Many thanks for your response.
I will try.