guino/BazzDoorbell #111: Trrying to make headway with this doorbell

djarvis, 2023-08-25:
Hi!

I got a cheap Tuya doorbell, manufacturer is Shenzhen Bilian Electric Co, model is 207-D.  I can't find much info about it.  I tried the SD card methods you described but nothing took.  My goal is to open an RTSP stream.

A port scan shows **21**/FTP, **23**/TELNET, and **6668**/Tuya TCP ports open.  No web ports like 80 or 8090 or anything like that.

The FTP port immediately spits back:
**inetd: can't execute /sbin/sftpd: No such file or directory**
and then disconnects.

The telnet port I can connect to but it immediately disconnects with no data.  An nmap scan shows busybox:
23/tcp    open          telnet       **BusyBox** telnetd 1.14.0 or later

The 6668 is the Tuya port and I can see various traffic on it when I do things like use TuyApi or some other app to connect.

There are around 20 UDP ports open.

Do you have any ideas or suggestions?  I have an extra one of these I'm willing to send to you if you want, or we can work together and I can learn a thing or two.

guino, 2023-08-28:
If it's a battery operated device I'd not even waste my time on it. If it's a hardwired device I'd say your best bet is to dump the firmware with a hardware programmer to look at it. 

Other than the above, if you can get the firmware version (using the phone app) it may be helpful so we can try a process (without hardware programmer) that is more similar to the firmware version on the device (there are several variations available).

I have no information on that device so it's hard to do anything without it in hands.

djarvis, 2023-08-28:
It is a hardwired device that hooks up to existing 24V.

The phone app does not indicate any sort of version, but I can see a UDP packet on powerup that suggests the firmware is 1.6.3.1:

{"cmdId":1,"tuya_uuid":"hwxd082741c1ca946c5e","tuya_pid":"18k9ap0dtnl6bvfk","devType":"Doorbell","devId":"","devName":"","netParam":{"isDHCP":1,"ip":"192.168.1.63","mask":"255.255.255.0","gate":"192.168.1.1","dns1":"","dns2":""},"httpPort":0,"httpsPort":0,"ddnsUrl":"","uid":"LBCS-000000-CCCJJXY","test_uid":"","token":"880e0d76","firmwareVer":"**1.6.3.1**"}

guino, 2023-11-13:
That firmware version is not similar to any of the devices I've seen in the past. You can try any of the processes I posted for 2.7.x, 2.9.x, 2.10.x and 4.x -- it is highly unlikely to break anything but chances are low that any of them will work either. Short of opening the device to read the flash with a hardware programmer or find rx/tx to mess with it, there's not much else we can do since telnet seems to just disconnect.