guino/BazzDoorbell #113: Creating a simple front-end for the doorbell for use with Nest Hub (or alike)

jilleb, 2023-10-13:
I am using the doorbell with my Athom Homey, and I cast a webpage to a Google Nest Hub whenever the doorbell is pressed, so I can see my doorbell cam stream pop on the screen:
[image]

To do so, I made a simple bit of HTML:
```html
<!DOCTYPE html>
<html>
<head>
    <style>
        body {
            margin: 0;
            padding: 0;
            overflow: hidden;
        }

        img {
            position: absolute;
            top: 0;
            left: 0;
            width: 100%;
            height: 100%;
        }
    </style>
</head>
<body>
    <img src="./cgi-bin/mjpeg.cgi" alt="Fullscreen Stream">
</body>
</html>

``` 

This will display the webcam image full screen. It will not be a moving image, but that was enough for a first proof of concept.
From here, I continued exploring what's possible, and I want to be able to trigger some commands on the doorbell, like:
- play sounds (I can't come to the door, I am not interested, and such)
- change day/night/auto mode
- reboot the doorbell
- other ideas are welcome.

So far, I came up with the following:
[image]

There's currently 1 issue: all the nightvision stuff needs user credentials to be sent with the GET request... which sucks because of the CORS stuff that doesn't allow you to fetch when you have credentials in a URL.  This isn't really my daily cup of tea, so if anyone has a brilliant idea how to get this working, let me know :-D

I tried creating a commands.cgi which will execute a wget server-side, but so far this wasn't succesful yet.

guino, 2023-11-13:
@jilleb you could potentially make a cgi file that changes the "night_mode" setting in /home/cfg/tuya_config.json and restarts ppsapp (so it reads it). Obviously this will disconnect the feed while it restarts but at least on my old 2.7.3 firmware it seems to have worked. The 'set' script can adjust the setting from command line if you wish to try it.

jilleb, 2023-11-14:
Hi @guino ,

I'll make it.
I am using the url endpoint to switch the night/day mode myself:

Http://admin:056565099@192.168.1.98/dsp/debug/xchdaynight/night/