guino/BazzDoorbell #114: Help needed with a Nedis WIFICDP10GY Doorbell

aldertsmits, 2023-10-15:
Hi,

I can't get the Nedis WIFICDP10GY doorbell to boot the sdcard.
Firmware version in Tuya app: 5.3.7

It's fully working in the Tuya App on my phone.
It got an open TCP/6668

I got the following from NMAP:
MAC Address: 90:31:4B:XX:XX:XX (Unknown)
Device type: general purpose
Running: Linux 4.X
OS CPE: cpe:/o:linux:linux_kernel:4.4.2
OS details: DD-WRT v3.0 (Linux 4.4.2)
Uptime guess: 0.016 days (since Sun Oct 15 15:16:28 2023)
Network Distance: 1 hop
TCP Sequence Prediction: Difficulty=262 (Good luck!)
IP ID Sequence Generation: All zeros

I tried multiple SD-cards, Formated them multiple different ways. (including convert to GPT and convert back to MBR)
I can't get a ping or telnet to open after reset with the SD-card in it.
I tried the 720p, 1080p and BazzDoorbell files on the sdcard.

Onvif device manager find a IPC_111641034 device on IP adres, but can't connect to port 8000/onvif/device_service

Any further suggestion to try?

guino, 2023-11-13:
I haven't seen that firmware version before -- very few 5.x firmware versions have worked with the existing processes in the past. It is a hard-wired device which most likely runs linux from your output, but they likely changed the bootloader/OS load address which would prevent any of the existing processes from working. Only way to make any progress would be to open it up and read the firmware with a hardware programmer or at least find/connect rx/tx to try to find what's changed.

th0ma7, 2024-02-21:
Guessing I'm facing a similar problem.  

**Model:** Geeni CS025 Wired Doorbell
**Firmware version:** V5.3.2

`nmap` output:
```
$ sudo nmap -O <IP>
Starting Nmap 7.94SVN ( https://nmap.org ) at 2024-02-20 20:17 EST
Nmap scan report for th0ma7-doorbell.th0ma7.lan (192.168.80.18)
Host is up (0.0051s latency).
Not shown: 999 closed tcp ports (reset)
PORT     STATE SERVICE
6668/tcp open  irc
MAC Address: 38:BE:AB:1D:99:56 (AltoBeam (China))
Device type: general purpose
Running: Linux 4.X
OS CPE: cpe:/o:linux:linux_kernel:4.4.2
OS details: DD-WRT v3.0 (Linux 4.4.2)
Network Distance: 1 hop

OS detection performed. Please report any incorrect results at https://nmap.org/submit/ .
Nmap done: 1 IP address (1 host up) scanned in 3.04 seconds
```

Sometimes the `nmap` varies and shows the following:
```
$ sudo nmap -O <IP>
Starting Nmap 7.94SVN ( https://nmap.org ) at 2024-02-20 20:21 EST
Nmap scan report for th0ma7-doorbell.th0ma7.lan (192.168.80.18)
Host is up (0.0029s latency).
Not shown: 999 closed tcp ports (reset)
PORT     STATE SERVICE
6668/tcp open  irc
MAC Address: 38:BE:AB:1D:99:56 (AltoBeam (China))
No exact OS matches for host (If you know what OS is running on it, see https://nmap.org/submit/ ).
TCP/IP fingerprint:
OS:SCAN(V=7.94SVN%E=4%D=2/20%OT=6668%CT=1%CU=44310%PV=Y%DS=1%DC=D%G=Y%M=38B
OS:EAB%TM=65D5503A%P=x86_64-pc-linux-gnu)SEQ(SP=F5%GCD=1%ISR=10D%TI=Z%CI=I%
OS:II=I%TS=7)OPS(O1=M5B4ST11NW3%O2=M5B4ST11NW3%O3=M5B4NNT11NW3%O4=M5B4ST11N
OS:W3%O5=M5B4ST11NW3%O6=M5B4ST11)WIN(W1=7120%W2=7120%W3=7120%W4=7120%W5=712
OS:0%W6=7120)ECN(R=Y%DF=Y%T=40%W=7210%O=M5B4NNSNW3%CC=Y%Q=)T1(R=Y%DF=Y%T=40
OS:%S=O%A=S+%F=AS%RD=0%Q=)T2(R=N)T3(R=N)T4(R=Y%DF=Y%T=40%W=0%S=A%A=Z%F=R%O=
OS:%RD=0%Q=)T5(R=Y%DF=Y%T=40%W=0%S=Z%A=S+%F=AR%O=%RD=0%Q=)T6(R=Y%DF=Y%T=40%
OS:W=0%S=A%A=Z%F=R%O=%RD=0%Q=)T7(R=Y%DF=Y%T=40%W=0%S=Z%A=S+%F=AR%O=%RD=0%Q=
OS:)U1(R=Y%DF=N%T=40%IPL=164%UN=0%RIPL=G%RID=G%RIPCK=G%RUCK=G%RUD=G)IE(R=Y%
OS:DFI=N%T=40%CD=S)

Network Distance: 1 hop

OS detection performed. Please report any incorrect results at https://nmap.org/submit/ .
Nmap done: 1 IP address (1 host up) scanned in 11.75 seconds
```

guino, 2024-03-05:
@th0ma7 the USB 4-pin reader is usually a UART adapter which can be helpful if it's TTL 3.3v but it won't read the flash. To read the flash you usually have to open the device remove the flash chip and use a flash programmer to read it. It's not something I'd recommend doing unless you have the right tools (heat gun + flash programmer) and some experience with them or you'll damage the board.

th0ma7, 2024-03-05:
Indeed sadly this isn't something i can easily do.  Still, I'll check if i can find someone in my surrounding who could help having the right tools.

Dr34mChas3r, 2025-03-06:
I Have alpina doorbell 8S  (v5.3.7) that have firmware  ppstrong-a5-tuya2_general-5.3.7.20221203 , and onvif works(can watch video without tuya(smartlife) app. Open ports: 8000, 8554, 6668. Can i have custom firmware anyone knows?

[image]

[image]

GFJHogue, 2025-04-02:
So I just started tinkering with my Energizer Doorbell cam (Bell 5S, v2.1), also on firmware `5.3.7`. Naturally I ended up here.

My initial runs of `nmap` looked exactly like yours, nothing but the `6668` port showing open.

Now the phone app I have it linked to is Energizer Connect. I'm not familiar enough to know how similar/different it is from the Tuya app, but my in-app device settings include an `Onvif Switch` toggle.
On its own, it didn't seem to do anything - no difference in the `nmap` open ports, ODM not detecting any camera on the network.

Then I recalled reading someone on here who managed to get ports opened up by changing some small thing... I honestly forget, but I figured I'd try flipping some more toggles in my EC app.
Literally the first setting I tried was disabling `Auto Upgrade` in the device firmware section, and behold:
```
PORT     STATE SERVICE
6668/tcp open  irc
8000/tcp open  http-alt
```

A second port opened up and ODM could find the camera!
I now have full access to the Onvif/RTSP details :) 
(I didn't have to do any of the steps laid out in the guides here)

I wonder if the Tuya app behaves similarly, or if others' cameras can be linked to a different app such as Energizer Connect?