guino/BazzDoorbell #117: Need Help With Generic Tuya Doorbell

LucasMedeiros-dev, 2024-02-25:
Hi, I've managed to access the u-boot and shell via uart from my generic tuya doorbell with rf433 chime.

[image]

Here's the log I've managed to extract.

I'd like some help on how to enable onvif & rtsp and how to make it persistent.

## Logs

"""sh
U-Boot 2010.06 (Aug 10 2022 - 17:37:59)

DRAM:  64 MiB

MMC:   

spi_flash_probe_default multi wire open flag is 0

*** Warning - bad CRC, using default environment

In:    serial

Out:   serial

Err:   serial

Net:   FH EMAC

Hit any key to stop autoboot:  1  0 

U-Boot> [1~getev-  nv

Unknown command '[1~getenv' - try 'help'

U-Boot> help

?       - alias for 'help'

arc_go  - start application at address 'addr'

base    - print or set address offset

bdinfo  - print Board Info structure

boot    - boot default, i.e., run 'bootcmd'

bootd   - boot default, i.e., run 'bootcmd'

bootm   - boot application image from memory

bootp   - boot image via network using BOOTP/TFTP protocol

chpart  - change active partition

cmp     - memory compare

coninfo - print console devices and information

cp      - memory copy

crc32   - checksum calculation

date    - get/set/reset date & time

dhcp    - boot image via network using DHCP/TFTP protocol

dma_cp  - dma memory copy

dump_phy_reg- dump phy reg
[... 232 lines cut ...]
fh-linux# cat /ho
fh-linux# cat /home/[J
fh-linux# cat /home/shellpid [J [J[J[J[J[J[J[J[J[J[J[J[J[J[J h
fh-linux# cat /home/[J
fh-linux# cat /home/shellpid [J
1
fh-linux# ac[J[Jcat/ [J[J /h
fh-linux# cat /home/[J[J[J[J[J[Jsu[J[Jus
fh-linux# cat /usr/[J h o [J[J[J
fh-linux# cat /usr/[J 
attr/    bin/     driver/  lib/     share/

fh-linux# cat /usr/[J
attr/    bin/     driver/  lib/     share/

fh-linux# cat /usr/[Jb
fh-linux# cat /usr/bin/[J 
EasyCam  RShell

fh-linux# cat /usr/bin/[J
EasyCam  RShell

"""

Sorry for the logging bugs.

guino, 2024-03-05:
@LucasMedeiros-dev looks like you already have root access - that's great!

On most of the newer devices you can enable onvif/rtsp by editing /home/cfg/tuya_config.json -- execute `cat /home/cfg/tuya_config.json` to see the file contents on your terminal.

You should be able edit the file on your computer with the following steps:
1-Insert SD card FAT32 formatted, power on device
2-on your shell (above), execute: `cp /home/cfg/tuya_config.json /mnt/mmc01/; sync`, power off device
3-Take the SD card out, put it on your computer and edit the tuya_config.json file to have onvif_enable set to 1, save/eject SD card
4-put the file back onto the device, power it on again
5-on your shell, execute: `cp /mnt/mmc01/tuya_config.json /home/cfg; sync`
6-reboot the device to use the new config file

It may be helpful to know what version of ppsapp you're running (usually shows in the phone app somewhere), you can alretnatively copy ppsapp to the SD card executing this  after step 1 above: `mount -t cramfs /dev/mtdblock5 /opt/pps; cp /opt/app/app* /mnt/mmc01/` -- the app.tar.* file should be in the root of the SD card (and ppsapp is inside of it), so feel free to post a zip of if for review and we can go from there.

LucasMedeiros-dev, 2024-03-17:
Hi! I got to work on the camera today, there's no CFG folder nor tuya_config.json, there's a folder named Tuya_config with the following files `log_seq_stat      tuya_enckey.db    tuya_user.db      tuya_user.db_bak`

Also found the command getVersion and it returns

```
************************************************
FW compile date =[May 27 2023 11:41:17]
  BUILD_FW_TYPE_VER: 1
  FW_VER_CODE      : 1631
  modelName        : D1
  customerName     : WXD
*************************************************

```

The mount command to mount ppsapp doesn't work

LucasMedeiros-dev, 2024-03-17:
After running something like get_burn_file, this file was present in the sdcard 

[Flash_ACdoorbell_D1_WXD_P20_V1.6.3.1_Tuya.zip](https://github.com/guino/BazzDoorbell/files/14628760/Flash_ACdoorbell_D1_WXD_P20_V1.6.3.1_Tuya.zip)

guino, 2024-05-07:
The zip file provided seems to be a full firmware dump file (8Mb). The main application seems to be 'EasyCam', and seems to be very different from the other 'ppsapp' files we've seen.

I opened the easeycam file in ghidra and the code did not seem to have any tuya_config.json references as you pointed out. It seems to be just reading any settings from the cloud server.
The only reference to RTSP features I found seemed to be initiated from the cloud (mqtt) server, but it didn't seem like there was anything that would initialize it, so doesn't seem like RTSP is available in any fashion.

This being a newer camera you may be able to use the WebRTC interface (https://www.reddit.com/r/smartlife/comments/oyqvdv/webrtc_stream_terminal_for_tuya_smartlife_cameras/) , if so you may be able to use https://www.scrypted.app/ to convert from WebRtc to RTSP, but that would be all.