guino/BazzDoorbell #118: Websockets?

oepi-loepi, 2024-03-07:
Would it be possible to add a websocket client or a websocket option to the ring? In that case it would be very convenient to have a websocket client to react in case of a push on the bell.

(I m developer of apps for the Dutch Eneco Toon Thermostat (rooted) and would like to rewrite my app to connect the doorbell to the toon eneco thermostat, which is also a kind of tablet hanging on the living room wall).

guino, 2024-03-07:
@oepi-loepi there are limited tools available in the device. As of right now `curl` and `nc` should be available in the busybox we install in the SD card. We have compiled some simple MQTT clients into some ARM devices, but this is the first time I hear a request for websockets.  Anything outside of these (which provide basic web/REST, raw TCP communication and MQTT notifications), you would either have to write + compile something to run in the device (likely using pure C and and the correct ARM cross compiler) OR you could use a 3rd device to fill the gap, for instance: you could have a basic application listening to a REST endpoint and making the same application perform any websocket communication you need (when the REST endpoint is requested using curl from the device). This application could be anywhere, like a spare machine in the network or the websocket server itself (at which point you could possibly perform whatever action from the REST endpoint request directly).

oepi-loepi, 2024-03-07:
Thanks… you are fast and precise.. love it.Yes, i now use homey to get the ring and then send the websocket. However since above 5000 users have the toon rooted, there is a mass request for a doorbell option. Most of them do not have seperate devices.(https://github.com/ToonSoftwareCollective)I will check the REST option. Do you have a small manual for that REST option?Kind regards, robVerstuurd vanaf mijn iPhone