guino/BazzDoorbell #119: Tuya V33.50.82

guino, 2024-06-14:
Hello There,
Any chance somebody tried/found a way to do this with a caméra cloud Tuya V33.50.82 (Kerui JS-P162) Version 33.50.82
[image]

Could be cool ! I baught 3 for 30 €. :-)

_Originally posted by @Rols38 in https://github.com/guino/BazzDoorbell/issues/2#issuecomment-2161107834_

guino, 2024-06-14:
@Rols38 I have not seen any device with that firmware version yet. Someone would have to try the available methods to see if anything works (doesn't seem likely without modifications) OR get UART access OR a dump of the firmware for review.

Rols38, 2024-06-15:
Thank you @guino for your feedback,

I would not try the avalable méthode yet due to the exotic version of this firmware. I like to play but i need to know that I can win first ! ;-)

In your opinion, Should I try to dump the firmware first or getting UART acccess is easyer ?
 
Does you or someone else can help me to do this or link me to the apropriate process ? 

Wich you all a grate day !

guino, 2024-06-20:
@Rols38 I have never seen a case where it 'hurt' the device by trying one of the existing methods. Most of the time it just doesn't do anything if it doesn't work.

UART access is usually easier because you just solder or insert some wires to the board but it hardly ever provides a way to root the device or dump the firmware. It most times gives some insight into the type of board/firmware and if it's running linux or something else.

Firmware dump usually provides a full insight into the device, but again no guarantee that it can be rooted without modifying/writing to the firmware directly. The problem with this option is that it usually requires removing the chip from the board (heat gun) -- I discourage anyone from trying to remove it with a soldering iron because it will most likely damage the board beyond any chance of repair.

Each option above requires specific and different hardware (UART-TTL adapter OR Flash programmer), so I'd start with what's available.