guino/BazzDoorbell #121: JPEG Address question

guino, 2024-06-20:
Hello @guino ,
 
I succeeded the previous steps but, at step 9, I don't understand where to find the jpeg address to adjust it in snap.cgi and mjpeg.cgi.

How to retrieve the jpeg address please?

**Thanks a lot for your work !**

_Originally posted by @Shadowws326 in https://github.com/guino/BazzDoorbell/issues/2#issuecomment-2177894545_

guino, 2024-06-20:
@Shadowws326 did you check https://github.com/guino/ppsapp-rtsp/issues/1 to see if your ppsapp has already been patched ? If so, your JPEG address should have been posted with the patch. If not, then you can post a zip with your `/home/app/ppsapp` (created in the SD card), along with the information from the /devices/deviceinfo URL so I can create a patch and post it along with the JPEG address for it.

Shadowws326, 2024-06-22:
Thank you @guino .

As you said, my ppsapp 21.10.5 was already in the list. Now MJPEG and snapshot works :)

I will continue today to try to activate the RSTP connection.