guino/BazzDoorbell #122: LSC Smart Connect doorbell - no Wifi

jankorr, 2024-08-21:
Hello. I've bought this device as "used". I was trying to connect it to my wifi network with Tuya Smart and  LSC Smart Connect Apps, but without sucess. Camera seeing QR Code, "beeps" 3 times, LED starts and continue blinking blue and ... nothing is happening, even after 2 minutes. When I unplug it, and plug it back again, starts normally (with short sound) and starts blink in red. I've checked motherboard - it's older version - "hackable" with markings: "Bell5s H1, 2.8.0.20200320". What do You think ? Firmware is broken? Should I make a flash copy and try to put other bin file (if I find it) ? Or maybe Wifi chip is dead? I did "reset" of device procedure (thru reset pin) already.

guino, 2024-08-22:
I have seen cameras with bad wifi chip (still have one that luckily also has ethernet), so it is possible it is just not able to connect to your wifi. I am assuming you haven't tried to do anything with it, but without wifi none of the hacks/mods we have will work. You could at most usa UART/TTL adapter and see what the log shows during boot up -- I am taking a guess it will just say it can't connect to the wifi for one reason or another.

Before you deep dive or throw it away, I would recommend trying to get it to connect to a cheap/dump router with only a basic 2.4Ghz wifi -- I have seen may cases of even robust and name brand products having issues connecting to fancy routers because of advanced settings and/or mix 5Ghz naming and such (xfinity/nest/bell routers specifically). I have also seen cases where a simple router reboot is enough to get devices connected (most recently with an amazon blink camera).

jankorr, 2024-08-30:
Guino, thank You for a advice. Apparently this was a problem with Wifi security (WPA2/WPA3 mixed together). In may LAN there is aroud 30 devices and only this doorbell had a problem with that setting. I've change it into WPA2 and started working properly.

KrChm, 2025-05-18:
Found this thread while have troubles with connection via qr code in Tuya App, i found walkaround my problem:
I made screenshoot of qr code, read this code and regenerated qr at some website, doorbell instantly conected to my Wifi.