guino/BazzDoorbell #124: After the new method of installing the hack my doorbell reboots at it self

alcomys, 2024-09-23:
After installing the hack my doorbell reboots at it self. 
My SD-Card has plenty of space so that is not full. 
Sometimes the doorbell works for a few hours but than  out of nowhere the doorbell is offline and reboots

devname   	"Smart Home Camera"
model	        "Bell 8S"
serialno	"xxxxxxxxxxxx"
softwareversion	"2.10.5"
hardwareversion	"BE8S_H1_V10_433"
firmwareversion	"ppstrong-c51-tuya2_lcs-2.10.5.20210806"

I hope someone can help me with this problem?

Thanks in advance.

alcomys, 2024-09-25:
Nobody have the same problem as i have?

Thanks

guino, 2024-10-01:
@alcomys this could be a hardware issue or SD card issue. Do you remember which method did you use to install the hack ? does the device boot and work normally without the SD card ?

You could also try a different SD card to be sure.

If you can post the result from [/proc/cmdline](http://admin:056565099@192.168.x.x/proc/cmdline) that could help as well (see note on 2.10.x at https://github.com/guino/BazzDoorbell/issues/2#issue-757550949 for steps to open port 80 to get the information) -- this is mostly to verify the hack was installed and which type was used.

alcomys, 2024-10-01:
> @alcomys this could be a hardware issue or SD card issue. Do you remember which method did you use to install the hack ? does the device boot and work normally without the SD card ?
> 
> You could also try a different SD card to be sure.
> 
> If you can post the result from [/proc/cmdline](http://admin:056565099@192.168.x.x/proc/cmdline) that could help as well (see note on 2.10.x at [#2 (comment)](https://github.com/guino/BazzDoorbell/issues/2#issue-757550949) for steps to open port 80 to get the information) -- this is mostly to verify the hack was installed and which type was used.

@guino 

Thank you for the replay.
I did that already, I changed the SD-card to a new one and the hack has been successful .
The problem i have with the reboots are only when i look at the RTSP stream via VLC player. 
So I can watch only an half hour then the camera reboots at it self.
When i use the camera normally via the tuya app, it works flawless. 
It goes wrong only when viewing via RTSP stream. 
The hack i used is the second hack with ppshack2 because the older method didn't work for me. 
This is my info from the command line.

devname   	"Smart Home Camera"
model	        "Bell 8S"
serialno	"xxxxxxxxxxxx"
softwareversion	"2.10.5"
hardwareversion	"BE8S_H1_V10_433"
firmwareversion	"ppstrong-c51-tuya2_lcs-2.10.5.20210806"
authkey	        "A4monXJ5kGMJ28Y5vB4uUKWL6QzOFqGA"
deviceid	"pp011a39cb12e986a06b"
identity	"MR2007120100114836"
pid	        "aaa"
WiFi MAC	"d4:d2:d6:b1:fb:55"

Thanks.

alcomys, 2024-10-01:
@guino 
I did zipped my files of the SD-CARD.

[Uploading SDCARD BACKUP.zip…]()

Thanks

guino, 2024-11-20:
@alcomys sorry for your issues -- I was unable to view the zip file (maybe I waited too long and it got removed?). The streaming using RTSP is different from the streaming using the app, it is possible the RTSP implementation in that specific firmware version is flawed/buggy and crashing after awhile. It is also possible to be a hardware issue.

Does the RTSP stream stop **exactly** after half hour (30 minutes) ? or does it stop at different intervals (i.e. 20 minutes, 40 minutes ? A hardware issue would likely happen at different intervals, a software issue would likely happen always at the same time.

I would also suggest trying the RTSP stream in a different application (not VLC) to see if the problem happens after the same amount of time or at all.

guino, 2024-11-20:
@alcomys you may also want to check for updates as there was a similar issue reported with another device and got fixed by an update -- see this thread: https://github.com/guino/LSCOutdoor1080P/issues/22#event-14249988869

alcomys, 2024-11-20:
> @alcomys sorry for your issues -- I was unable to view the zip file (maybe I waited too long and it got removed?). The streaming using RTSP is different from the streaming using the app, it is [...]

@Guino,

Thanks for replay. 
I did try different video players, The RTSP stream stops every 30 minutes so it looks like it needs some of reset or it has some timer somewhere inside the software. 
But when using in original state it is working without any problem also view the video with the tuya app works fine. 

About the update, I have the latest update available from tuya installed on my doorbell.
But I did removed the hack now because of the non working camera. 

Thanks