guino/BazzDoorbell #125: IP CAMERA APTI-YK21-TUYA

blahlt, 2025-02-08:
Hi. I have "IP CAMERA APTI-YK21-TUYA / 2.4G Wi-Fi Smart Door Camera". None of the methods works for me :( Any ideas where I can find UART pins? Maybe someone has ANYKA AK3918EN080 V330L datasheet?
[pcb bottom](https://i.imgur.com/rbKs5QI.jpeg)
[pcb top](https://i.imgur.com/WEnIIjy.jpeg)

guino, 2025-02-26:
RX/TX could be these 2 round connection points in the middle (and you'd use GND from anywhere else):

[image]

I don't see any other obvious locations -- if this is a battery operated device, there are good chances it may be running RTOS and not linux.

blahlt, 2025-03-07:
Yes, you are right!
[Image](https://github.com/user-attachments/assets/a07e6987-fd64-4eaa-86ff-31cff8c3ffcb)
Also this camera have telnet, video stream and so on..
Thanks!