guino/BazzDoorbell #126: hack stops working after about a week

bkbartk, 2025-03-09:
I use the bazzdoorbell hack for a couple of years now, but since around October I have a lot of issues.

- telnet keeps working,
- I'm able to make snapshots.
- 433mhz I use as a fallback still works (probably irrelevant to the issue).
- but the RSTP stream stops,
- no new recordings are craeted on the SD card, there is plenty of space on the SD card.
- none of my MQTT messages work.

it's like not everything is loaded as expected.
I had the same behaviar twice when my SD card was full. (auto clean failed because of a corrupt file)
I replaced my SD card twice, last time using a class A2 model.

reboot not always help,
a power cycle does now. but also not always.

is anyone familiar with this behaviar? is my camera toast and do I need to acquire a new one?

Nigel1992, 2025-03-30:
This might sound obvious, but I'm still gonna ask it.
Did you update the firmware of your camera ?
If so, this could patch/brake thing.

bkbartk, 2025-03-30:
I did not. 
But it looks like enabling cloud access again makes the issue go away. I still want to test a little longer before I can be certain this is the issue. 

My device needs a cloud connection. Else it won’t boot. After booting I can kill the cloudconnection. 
I don’t know how it works. Maybe I get a token witch bos is valid only 7 days and before infinite. 

But as mentioned. Unwanted to test some more and if I have any real results I will give an update.

bkbartk, 2025-04-08:
I haven't had any issues since I allow my doorbell to connect to the internet all the time,
So the issue seems to be related to that,
not ideal but I will live with it.